ucb-bar/chisel2-deprecated #358: Unintuitive width inference/Mux/SInt behavior: sign-extension fails in presence of a UInt

ccelio, 2015-01-30:
````scala
    val out = UInt(width=8)
    val b = UInt(3, width=2)
    out := Mux(condition, UInt(0), b.SInt())
````

If condition is false, `out == 3`, not `0xFF`.  Intuitively, I am expecting b.SInt() to sign-extend to the full width of the LHS variable `out`. 

````Verilog
    // Verilog
    wire[7:0] T2;
    wire[1:0] T1; 
    wire[1:0] b;
    … 
    assign T2 = {6'h0, T1};
    assign T1 = condition ? 2'h0 : b;       
````                              

Instead, the Mux() becomes a Node of the length of the maximum child member (2-bits in this case). Sign-extension is only performed up to the width of the Mux().  Then, the Mux is zero-extended to the length of the LHS of the assignment ( `out = Mux()` ). 

The above code can be fixed by using a SInt(0), which apparently tells the Mux() "No really, I want you to be a sign-extending Node".  By default, Mux's are otherwise unsigned. 

Looking through Rocket code, I find the following example (simplified) from csr.scala:

````scala
   val io_evec = UInt(width = VaddrBits+1)
   val reg_evec = Reg(Bits(width = VaddrBits))
   val reg_epc = Reg(Bits(width = VaddrBits+1))
   io_evec := Mux(exception, reg_evec.toSInt, reg_epc).toUInt
````
This works currently because the final width of the `Mux()` and the LHS variable `io_evec` are the same (VaddrBits+1).  The `reg_evec` variable is properly sign-extended to the width of the Mux, and there is no final zero-padded extension to perform that would raise an error. 

**Solutions?**

One solution is to throw an error if a user mixes signed and unsigned types in their Muxes (at least, if the Mux width and the LHS width are mismatched and extension is required). I don't think that this adds much in the way of superfluous casting of nodes and will save users from the occasional edge-case of unintuitive width inference behavior.

ccelio, 2015-01-30:
Another solution is to make muxes SInt and not UInt in the presence of SInts, as SInt is the more general type.

In general, we should probably do whatever C does when it comes to mixing widths and signed-ness.

sdtwigg, 2015-02-04:
Fixed. Warning is currently emitted in this case.

closed 2015-02-04 by sdtwigg