ucb-bar/chisel2-deprecated #359: Weird compiler error for write-masked Mems in Verilog backend

zhemao, 2015-02-02:
While working on ZScale, I encountered a compiler failure when trying to elaborate for the FPGA. The problem is on the following line.

https://github.com/ucb-bar/chisel/blob/master/src/main/scala/Verilog.scala#L315

The line fails if the UInt has no inputs. Changing this to

    if (x.inputs.length > 0) find_gran(x.inputs(0)) else 1

seems to fix the problem.

sbeamer, 2015-02-02:
Could you provide a pointer to the code that exposes this? Having 0 inputs within that part of find_gran is problematic.

zhemao, 2015-02-03:
https://github.com/ucb-bar/reference-chip/tree/io-coproc

It works fine in the other backends. It's only a problem when trying to use non-inlined mems in the verilog backend.

sbeamer, 2015-02-04:
The above commit should fix this crash as well as the granularity being incorrectly inferred. Please confirm.

zhemao, 2015-02-04:
Yeah, it works. What should we do about the chisel submodule in reference-chip, should we sync it with master?

closed 2015-02-04 by zhemao