ucb-bar/chisel2-deprecated #374: Multiply instantiated modules in Chisel result in redundant (functionally equivalent) module definitions in generated Verilog

rimasaviz, 2015-03-04:
When instantiating multiple instances of a module in a design the generated Verilog may contain multiple, redundant module definitions which are each instantiated once.  For example, pulling the current reference-chip design and building a two core/tile configuration results in verilog that looks like:

cd vsim
make CONFIG=HwachaMesiL2T2B8C512Config

looking in generated-src/Top.HwachaMesiL2T2B8C512Config.v we see these two module definitions:

module RocketTile_0(input clk, input reset, ... )
module RocketTile_1(input clk, input reset, ... )

which are later instantiated as follows:

RocketTile_0 RocketTile_0(.clk(clk), .reset(resetSigs_0), ... )
RocketTile_1 RocketTile_1(.clk(clk), .reset(resetSigs_0), ... )

This is bad, because the two tile modules are identical and we are depriving the synthesis tool of this information (resulting in longer runtimes and worse QoR).  This is particularly troublesome in the case where we want to place and route the RocketTile once and then instantiate two of them at the top level - since the modules have different names the tool reasonably assumes they aren't identical.  I have been manually hacking the generated verilog to look like:

module RocketTile(input clk, input reset, ... )
 
RocketTile RocketTile_0(.clk(clk), .reset(resetSigs_0), ... )
RocketTile RocketTile_1(.clk(clk), .reset(resetSigs_0), ... )

which makes it possible to use the hierarchical flow (though other submodules in the design still have this issue).  A diff of the generated verilog for modules RocketTile_0 and RocketTile_1 is attached - the only difference appears to be in the naming of internal nets and ports on modules.

1c1
< module RocketTile_0(input clk, input reset,
---
> module RocketTile_1(input clk, input reset,
129,184c129,184
<   wire TileLinkIOWrapper_0_io_in_acquire_ready;
<   wire TileLinkIOWrapper_0_io_in_grant_valid;
<   wire[3:0] TileLinkIOWrapper_0_io_in_grant_bits_header_src;
<   wire[3:0] TileLinkIOWrapper_0_io_in_grant_bits_header_dst;
<   wire[1:0] TileLinkIOWrapper_0_io_in_grant_bits_payload_addr_beat;
<   wire[127:0] TileLinkIOWrapper_0_io_in_grant_bits_payload_data;
<   wire[2:0] TileLinkIOWrapper_0_io_in_grant_bits_payload_client_xact_id;
<   wire[2:0] TileLinkIOWrapper_0_io_in_grant_bits_payload_manager_xact_id;
<   wire TileLinkIOWrapper_0_io_in_grant_bits_payload_is_builtin_type;
<   wire[2:0] TileLinkIOWrapper_0_io_in_grant_bits_payload_g_type;
<   wire TileLinkIOWrapper_0_io_in_finish_ready;
<   wire TileLinkIOWrapper_0_io_out_acquire_valid;
<   wire[3:0] TileLinkIOWrapper_0_io_out_acquire_bits_header_src;
<   wire[3:0] TileLinkIOWrapper_0_io_out_acquire_bits_header_dst;
<   wire[25:0] TileLinkIOWrapper_0_io_out_acquire_bits_payload_addr_block;
<   wire[2:0] TileLinkIOWrapper_0_io_out_acquire_bits_payload_client_xact_id;
<   wire[1:0] TileLinkIOWrapper_0_io_out_acquire_bits_payload_addr_beat;
<   wire[127:0] TileLinkIOWrapper_0_io_out_acquire_bits_payload_data;
<   wire TileLinkIOWrapper_0_io_out_acquire_bits_payload_is_builtin_type;
<   wire[2:0] TileLinkIOWrapper_0_io_out_acquire_bits_payload_a_type;
<   wire[16:0] TileLinkIOWrapper_0_io_out_acquire_bits_payload_union;
<   wire TileLinkIOWrapper_0_io_out_grant_ready;
<   wire TileLinkIOWrapper_0_io_out_finish_valid;
<   wire[3:0] TileLinkIOWrapper_0_io_out_finish_bits_header_src;
<   wire[3:0] TileLinkIOWrapper_0_io_out_finish_bits_header_dst;
<   wire[2:0] TileLinkIOWrapper_0_io_out_finish_bits_payload_manager_xact_id;
<   wire TileLinkIOWrapper_0_io_out_probe_ready;
<   wire TileLinkIOWrapper_0_io_out_release_valid;
<   wire TileLinkIOWrapper_1_io_in_acquire_ready;
<   wire TileLinkIOWrapper_1_io_in_grant_valid;
<   wire[3:0] TileLinkIOWrapper_1_io_in_grant_bits_header_src;
<   wire[3:0] TileLinkIOWrapper_1_io_in_grant_bits_header_dst;
<   wire[1:0] TileLinkIOWrapper_1_io_in_grant_bits_payload_addr_beat;
<   wire[127:0] TileLinkIOWrapper_1_io_in_grant_bits_payload_data;
<   wire[2:0] TileLinkIOWrapper_1_io_in_grant_bits_payload_client_xact_id;
<   wire[2:0] TileLinkIOWrapper_1_io_in_grant_bits_payload_manager_xact_id;
<   wire TileLinkIOWrapper_1_io_in_grant_bits_payload_is_builtin_type;
<   wire[2:0] TileLinkIOWrapper_1_io_in_grant_bits_payload_g_type;
<   wire TileLinkIOWrapper_1_io_in_finish_ready;
<   wire TileLinkIOWrapper_1_io_out_acquire_valid;
<   wire[3:0] TileLinkIOWrapper_1_io_out_acquire_bits_header_src;
[... 857 lines cut ...]
>        .io_out_acquire_bits_payload_a_type( TileLinkIOWrapper_3_io_out_acquire_bits_payload_a_type ),
>        .io_out_acquire_bits_payload_union( TileLinkIOWrapper_3_io_out_acquire_bits_payload_union ),
>        .io_out_grant_ready( TileLinkIOWrapper_3_io_out_grant_ready ),
2308,2312c2308,2312
<        .io_out_finish_valid( TileLinkIOWrapper_1_io_out_finish_valid ),
<        .io_out_finish_bits_header_src( TileLinkIOWrapper_1_io_out_finish_bits_header_src ),
<        .io_out_finish_bits_header_dst( TileLinkIOWrapper_1_io_out_finish_bits_header_dst ),
<        .io_out_finish_bits_payload_manager_xact_id( TileLinkIOWrapper_1_io_out_finish_bits_payload_manager_xact_id ),
<        .io_out_probe_ready( TileLinkIOWrapper_1_io_out_probe_ready ),
---
>        .io_out_finish_valid( TileLinkIOWrapper_3_io_out_finish_valid ),
>        .io_out_finish_bits_header_src( TileLinkIOWrapper_3_io_out_finish_bits_header_src ),
>        .io_out_finish_bits_header_dst( TileLinkIOWrapper_3_io_out_finish_bits_header_dst ),
>        .io_out_finish_bits_payload_manager_xact_id( TileLinkIOWrapper_3_io_out_finish_bits_payload_manager_xact_id ),
>        .io_out_probe_ready( TileLinkIOWrapper_3_io_out_probe_ready ),
2319c2319
<        .io_out_release_valid( TileLinkIOWrapper_1_io_out_release_valid )
---
>        .io_out_release_valid( TileLinkIOWrapper_3_io_out_release_valid )
2331,2332c2331,2332
<     assign TileLinkIOWrapper_1.io_in_acquire_bits_header_src = {1{$random}};
<     assign TileLinkIOWrapper_1.io_in_acquire_bits_header_dst = {1{$random}};
---
>     assign TileLinkIOWrapper_3.io_in_acquire_bits_header_src = {1{$random}};
>     assign TileLinkIOWrapper_3.io_in_acquire_bits_header_dst = {1{$random}};

ucbjrl, 2015-03-10:
When I try to duplicate this, instead of "TileLinkIOWrapper_x", I have "HeaderlessTileLinkIOWrapper_x" and the two sub-modules (HeaderlessTileLinkIOWrapper_1 and HeaderlessTileLinkIOWrapper_0) appear to be different. Could you verify that your TileLinkIOWrapper_0 and TileLinkIOWrapper_1 modules are identical?

ucbjrl, 2015-03-10:
Sorry, in your case it looks like the common sub-modules are TileLinkIOWrapper_0 and TileLinkIOWrapper_2 and they do look identical. I'll take a closer look.

ucbjrl, 2015-03-13:
I believe I have a fix for this with PR #381.

ucbjrl, 2015-03-31:
Should be fixed by Stephen's Verilog fixes of March 18 and 19 (2015).

closed 2015-03-31 by ucbjrl