ucb-bar/chisel2-deprecated #379: Build system and README

sclukey, 2015-03-11:
I recently tried to build chisel from its source and I found several issues that would like to bring up.

First of all, there are no obvious build instructions. Generally "how to build" should come before "what to do once you've built", so I found the "Chisel developers" section or the readme somewhat confusing. I would suggest that the build instructions be first in that section.

Once in the build section, the `mvn install` does not work unless the Oscar libs are installed, but the `./install_maven_libs` command doesn't work because the libraries (in `./lib`) were removed in 6bdda0450446d5efedf841ca2d147910d5b3d445 so I had to revive them from the history.

Next, the `sbt scct:test` command doesn't work because scct is no longer used (though that was pretty recent, 2957ac51abeed645e739a296f1000d997fb2d1d6 ).

Finally, the `./configure` script doesn't work and seems useless because the project was changed from sbt to maven quite some time ago.

Overall the readme and build system just seem very out of date, so I'm hoping they can be updated.

ucbjrl, 2015-03-11:
> Once in the build section, the mvn install does not work unless the Oscar libs are installed, but the ./install_maven_libs command doesn't work because the libraries (in ./lib) were removed in 6bdda04 so I had to revive them from the history.

Steven, thanks for bringing this to our attention. You are correct. The README is woefully out of date.

We will address this in the very near future.

In the meantime, you should be able to build and install a local version of Chisel by issuing the following command from the chisel project directory (assuming you have sbt, scala and make installed):

% make clean publish-local

You can run the tests using:

% make test

Please give this a try and report any problems you experience. We welcome your feedback.

sclukey, 2015-03-11:
Aha, I didn't notice the makefile somehow. Those commands work, thanks.

rupertlssmith, 2015-03-11:
Hi Steven,

I contributed the Maven build and that ./install_maven_libs script a while
back, because Maven is quite standard in the open source Java world. I do
think its nice if a Java (and I'm including Scala when I say Java) project
can hook its dependencies up to the Maven eco-system, and publish its build
artifacts on Maven central. However, I can understand why those oscar libs
would get removed; they are binaries not source. Its a shame they even had
to be put there in the first place, but oscar .jars are not in Maven
central either so there was not much option, other than writing to the
oscar devs and encouraging them to publish to Maven central, and then
waiting for that to happen.

Perhaps replacing that script with one that uses 'wget' to download the
oscar libs might work? To get them in the first place I had to search for
them and then figure out which version etc.

OTOH, maybe the maven and sbt scripts should be dropped and build
instructions standardized on make, if that is the main tool in use to build
it now.

Rupert

sclukey, 2015-03-12:
Rupert,

I agree that having Chisel in Maven central is very useful, and that support should be kept without being intrusive to the normal build method (apparently `make` calling `sbt`). As for Oscar, honestly I'm not sure that it is even needed anymore. I removed the Oscar dependencies from the `pom.xml` file and `mvn install` ran just fine. I guess whatever used Oscar is gone, so maybe just update the pom file then there is no need to have the jar files or the `./install_maven_libs` script.

As for actually building, I like the current setup of make wrapping calls to sbt. Make is simple and commonly understood, and since sbt used in the chisel tutorials and elsewhere I don't see why chisel itself would randomly switch to maven or raw scalac. That seems to me to be an unnecessary complication.

rupertlssmith, 2015-03-12:
On 12 March 2015 at 01:27, Steven Clukey <notifications@github.com> wrote:

> sbt used in the chisel tutorials and elsewhere I don't see why chisel
> itself would randomly switch to maven or raw scalac. That seems to me to be
> an unnecessary complication.
>

Yes, only reason I created a Maven script is so that it would push the
chisel artifacts to my local repository under ~/.m2/repository. Then I had
another project which consumed the build artifacts from there.

If I knew sbt, perhaps I could have achieved that using it, and never
ventured into Maven land. It certainly seems to be possible:

http://www.scala-sbt.org/0.12.1/docs/Detailed-Topics/Publishing.html

Rupert

ucbjrl, 2015-03-12:
On Mar 12, 2015, at 1:51 AM, Rupert Smith <notifications@github.com> wrote:

> On 12 March 2015 at 01:27, Steven Clukey <notifications@github.com> wrote:
>
>
> Yes, only reason I created a Maven script is so that it would push the
> chisel artifacts to my local repository under [...]

This is, in fact, how we currently publish Chisel updates

rupertlssmith, 2015-03-12:
On 12 March 2015 at 16:04, Jim Lawson <notifications@github.com> wrote:

> On Mar 12, 2015, at 1:51 AM, Rupert Smith <notifications@github.com>
> wrote:
>
> wrote:
> to be
> had
>
> This is, in fact, how we currently publish Chisel updates

You should probably just delete the Maven build in that case, as it is now
redundant. Will be less confusing for newcomers if it is not there.

Rupert

ucbjrl, 2015-03-12:
I'm marking this issue closed (pr #380). Feel free to re-open it or submit a create a new issue if you have more suggestions.

closed 2015-03-12 by ucbjrl