ucb-bar/chisel2-deprecated #384: Data width lost in Vec

mzimmer, 2015-03-20:
I can't get a Vec working as a ROM. I'm sure I could work around it, but this doesn't seem like correct behavior. Is there a bug, is the documentation out-of-date, or am I doing something wrong?

Consider this example code:
```scala
class HelloModule extends Module {
  val io = new Bundle {
    val in = SInt(INPUT, 32)
    val index = UInt(INPUT, 2)
    val out = SInt(OUTPUT, 32)
  }
  val rom = ...
  io.out := io.in - rom(io.index)

}
```

According to the 2.2.0 Tutorial, I can do something like this:
```scala
val rom = Vec((0 until 4).map(i => SInt(i, width = 32))) 
```
But this gives an error
`"SInt.matchWidth with unknown width: 32, node /*? in class HelloModule*/ Chisel.SInt(OUTPUT, width=None, connect to 1 inputs: (0x0[Chisel.Literal] in HelloModule)) in class HelloModule"`

According to the 2.2.0 Manual, I could also do this:
```scala
  val rom = Vec(SInt(0), SInt(1), SInt(2), SInt(3)) { SInt (width = 32) }
```
But a different error
```
overloaded method value apply with alternatives:
[error]   (ind: Chisel.UInt)Chisel.SInt <and>
[error]   (idx: Int)Chisel.SInt <and>
[error]   (name: String)Chisel.Data
[error]  cannot be applied to (Chisel.SInt)
[error]   val rom = Vec(SInt(0), SInt(1), SInt(2), SInt(3)) { SInt (width = 32) }
[error]                ^
[error] one error found
```
And I'm back to the first error if I try:
```scala
val rom = Vec(SInt(0, 32), SInt(1, 32), SInt(2, 32), SInt(3, 32))
```

Ideas?

mzimmer, 2015-03-20:
As always, I ran into the problem using latest.release and forgot to check if it's actually problem in the latest github. I'll check...

ucbjrl, 2015-03-23:
This is a bug in SInt.matchWidth(). A fix is forthcoming.

ucbjrl, 2015-03-25:
We need to update the documentation as well. It reflects old usage patterns.