ucb-bar/chisel2-deprecated #388: Verilog backends generates "dead" nodes

boehmerst, 2015-03-31:
As soon as the chisel code includes registers the Verilog backend generates unused wires or a chain of unused wires respectively (see the commented assignments of the generated code below). This is for sure not critical but it bloats the generated code. It seems like filtering the wires to have more than one consumer within the backend catches all the unused wires but is hacky of course.

- Stephan
..........................................................................................................................................................
..........................................................................................................................................................
class RegTest extends Module {
  val io = new Bundle {
    val inc   = Bool(INPUT)
    val dec   = Bool(INPUT)
    val clear = Bool(INPUT)
    val out1  = UInt(OUTPUT, 16)
  }
  val res1 = Reg(init = UInt(0, 16))

  res1 := ~res1
  when(io.inc) {
    res1 := res1 + UInt(1)
  }.elsewhen(io.dec) {
    res1 := res1 - UInt(1)
  }.elsewhen(io.clear) {
    res1 := UInt(0)
  }
  io.out1 := res1
}
...........................................................................................................................................................
...........................................................................................................................................................
module RegTest(input clk, input reset,
    input  io_inc,
    input  io_dec,
    input  io_clear,
    output[15:0] io_out1
);

  reg [15:0] res1;
  //wire[15:0] T11;
  //wire[15:0] T0;
  //wire[15:0] T1;
  //wire[15:0] T2;
  wire[15:0] T3;
  wire[15:0] T4;
  wire[15:0] T5;
  wire T6;
  wire T7;
  wire T8;
  wire T9;
  wire T10;

`ifndef SYNTHESIS
// synthesis translate_off
  integer initvar;
  initial begin
    #0.002;
    res1 = {1{$random}};
  end
// synthesis translate_on
`endif

  assign io_out1 = res1;
  //assign T11 = reset ? 16'h0 : T0;
  //assign T0 = T8 ? 16'h0 : T1;
  //assign T1 = T6 ? T5 : T2;
  //assign T2 = io_inc ? T4 : T3;
  assign T3 = ~ res1;
  assign T4 = res1 + 16'h1;
  assign T5 = res1 - 16'h1;
  assign T6 = T7 & io_dec;
  assign T7 = io_inc ^ 1'h1;
  assign T8 = T9 & io_clear;
  assign T9 = T10 ^ 1'h1;
  assign T10 = io_inc | io_dec;

  always @(posedge clk) begin
    if(reset) begin
      res1 <= 16'h0;
    end else if(T8) begin
      res1 <= 16'h0;
    end else if(T6) begin
      res1 <= T5;
    end else if(io_inc) begin
      res1 <= T4;
    end else begin
      res1 <= T3;
    end
  end
endmodule

sdtwigg, 2015-04-01:
This is known and somewhat intentional.

To improve readability, if a mux tree is used to feed a register, then the mux tree is replicated as if statements inside the assign statement. The mux tree is still emitted as wire+assigns in case other logic uses part of the mux tree.

Due to the complexity of the backend rewrites required to prune these before Verilog emission, fixing this behavior is deferred until the Chisel 3.0 major rewrite. In the interim, synthesis tools should prune the dead nodes anyway.

SebastianBoe, 2015-10-20:
+1

Readability of the verilog code is important when debugging, and not having dead code helps towards that. Looking forward to Chisel 3.0.