ucb-bar/chisel2-deprecated #397: Do blackboxes really need to name all IO pins with "io_"?

ben-k, 2015-04-16:
Let's say I have a piece of Verilog IP described thusly:

```verilog
module my_ip(
  input my_input,
  output my_output
);
  // Analog or otherwise non-Chisel-y functionality; port names can't be modified!
endmodule
```

I'd love to instantiate this in Chisel using BlackBox, but the best I can do is:

```scala
class my_ip extends BlackBox {
  val io = new Bundle {
    val my_input = Bool(INPUT)
    val my_output = Bool(OUTPUT)
  }
}
val ip = new my_ip
```

The generated Verilog unfortunately then looks like this:

```verilog
my_ip ip(
  input io_my_input,
  output io_my_output
);
```

Which doesn't match the module definition.  Is there an (undocumented) way to make the Verilog generated from BlackBox IOs not start with the string "io_"?  Otherwise you can't really plug in arbitrary black-boxed modules in the way that I think is intended.

CuppoJava, 2015-04-16:
We're working on that feature for Chisel 3.0. Perhaps someone else knows of
a magic workaround for getting it to work in Chisel 2.
  -Patrick

albert-magyar, 2015-04-16:
In Chisel 2, you can set the name (as emitted by a backed) of any wire to be whatever you like by invoking setName(name: String).

ben-k, 2015-04-16:
Thanks, I'll demote this to a documentation request.