ucb-bar/chisel2-deprecated #406: Literal folding does not respect specified widths

sdtwigg, 2015-05-04:
For:
```scala
  io.out := UInt(1,1) + UInt(1,1)
```
is equivalent to
```scala
  io.out := UInt(2,8)
```

However, literal folding is not always that intelligent. e.g.
```scala
  val y = UInt(); y := UInt(1,1)
  io.out := y + y
```
is equivalent to
```scala
  io.out := UInt(0,8)
```

Technically, the latter behavior is most proper and the former behavior is incorrect. The + operation is specified to not increase the width. Literal folding across + should itself also not increase the width. To address possible user error, a warning should be emitted that a folded literal result did not fit into the specified width.

Second note:
If the latter case is changed slightly to y := UInt(1) (and thus the width has not been specified) then nothing changes in the final result. This is probably still correct; however, there is no proper specification of what happens when a literal width is not specified (particularly, when the width should be assumed to be computed). Thus, it is trickier to discuss this case without enforcing a specification.

aswaterman, 2015-05-04:
> 
> For:
> 
>   io.out := UInt(1,1) + UInt(1,1)
> is equivalent to
> 
>   io.out := UInt(2,8)
> However, literal folding is not always that intelligent. e.g.
> 
>   val y = UInt(); y := UInt(1,1)
>   io.out := y + y
> is equivalent to
> 
>   io.out := UInt(0,8)
> Technically, the latter behavior is most proper and the former behavior is incorrect. The + operation is specified to not increase the width. Literal folding across + should itself also not increase the width. To address possible user error, a warning should be emitted that a folded literal result did not fit into the specified width.
I changed literal folding to do the second thing a while ago, but someone stomped on my changes and I didn't care enough to revisit it.

ucbjrl, 2015-05-04:
Could it have been one of these commits?
 - Don't lose bits when combining literals.
 - Don't override an explicitly set width for Literals.

ucbjrl, 2015-05-06:
This should be closed by #408.

closed 2015-05-06 by ucbjrl