ucb-bar/chisel2-deprecated #407: Don't care literal ("?

sdtwigg, 2015-05-04:


ucbjrl, 2015-05-04:
Should we outlaw the use of "?" in Literals in anything other than comparison ("===") operators? If not, what are the rules for combining "don't cares"?

ucbjrl, 2015-05-06:
This should be closed by #407.

closed 2015-05-06 by ucbjrl

David-Greaves-UoC-CL, 2015-06-12:
Sorry if this reopens a closed issue, but I wonder if people tend to forget the important role of don't cares in logic synthesis?  We teach first year students all about Karnaugh maps, but then forget to assign don't care in switch statements for the don't care cases.  Chances for logic optimisation are then lost.   So I'd be interested to know whether assigning a '?' in the otherwise case of a Chisel switch is recommended and whether it propagates through to the generated RTL so the back end tools do the right thing.

aswaterman, 2015-06-12:
We should exploit don't-cares.  The difficulty is the conflation of
don't-care with don't-know.  Verilog x-behavior is sufficiently pernicious
that we have shunned its omission for simulation purposes.

In some Chisel microprocessor implementations, we have automatically
invoked logic minimization tools at elaboration time, getting the bulk of
the benefit of the x without the possibility that the synthesis result
exhibits different behavior than the simulation model.  We'd like to
generalize this into a facility that others can take advantage of.