ucb-bar/chisel2-deprecated #412: JVM out of memory when running sbt for rocket-chip

richardarase, 2015-05-05:
richard@richard-HP-ENVY-15-Notebook-PC:~/Rocket/rocket-chip$ cd emulator richard@richard-HP-ENVY-15-Notebook-PC:~/Rocket/rocket-chip/emulator$ make debug cd /home/richard/Rocket/rocket-chip && java -Xmx2048M -Xss8M -XX:MaxPermSize=128M -jar sbt-launch.jar "project rocketchip" "elaborate Top --noIoDebug --backend c --configInstance rocketchip.DefaultCPPConfig --targetDir emulator/generated-src-debug --debug --vcd --ioDebug" sbt appears to be exiting abnormally. The log file for this session is at /tmp/sbt8961713207461919529.log java.lang.OutOfMemoryError: Java heap space Error during sbt execution: java.lang.OutOfMemoryError: Java heap space make: *** [generated-src-debug/Top.DefaultCPPConfig.h] Error 1

ben-k, 2015-05-05:
The typical fix to this is to increase the maximum heap size allocated to Java, e.g. by changing the argument to -Xmx4G.  You can make this change in the top-level Makefrag.

That said, 2GB is usually enough for this, so perhaps something else is going on.

sdtwigg, 2015-05-10:
Do you have the same issue for non-debug compilations?

Debug mode does increase the build complexity slightly (although usually moreso in compilation time, not required memory)

ccelio, 2015-05-10:
Jim has an incoming fix that should help greatly (by not spitting out two copies of the .cpp files).

richardarase, 2015-05-10:
Yes, the same issue for non debug compilation.

Richard