ucb-bar/chisel2-deprecated #414: Instance naming from Vecs could be saner

ben-k, 2015-05-05:
Instance names generated by Vecs tend to look like this:

```scala
instance
instance_1
instance_2
...
```

From a scripting standpoint, it would be great if they could instead look like this:
```scala
instance_0
instance_1
instance_2
```

albert-magyar, 2015-05-06:
Can you provide example code? I tried reproducing it, but got something like `instance_0` in my output.

ben-k, 2015-05-07:
Try building a two-core version of Rocket Chip.  When I do this, the Tile instances are named RocketTile and RocketTile_1.

If the behavior doesn't reproduce for you, then I am probably using an outdated version of Chisel, in which case feel free to close this.

sdtwigg, 2015-05-13:
I suspect this is from: https://github.com/ucb-bar/chisel/blob/master/src/main/scala/Backend.scala#L140
Fixing this issue would require a rewrite of the naming logic to backtrack once a collision was found and assign the first user of the name (instance in the supplied example) to an iterated name (e.g. instance_0). This adjustment would cause a corresponding weirdness; however, when the names just happen to collide and the associated items are not related at all.

Alternatively, you could more explicitly name instances that you know are part of a set (or the code that tries to find names for these things could be a little more intelligent and attempt traversing Iterables).