ucb-bar/chisel2-deprecated #439: Chisel somehow finds a way to build negative width wires

ben-k, 2015-06-17:
The ReferenceChip FlowThroughSerializer class in uncore/src/.../util.scala is this:

```scala
class FlowThroughSerializer[T <: HasTileLinkData](gen: T, n: Int) extends Module {
  val io = new Bundle {
    val in = Decoupled(gen.clone).flip
    val out = Decoupled(gen.clone)
    val cnt = UInt(OUTPUT, log2Up(n))
    val done = Bool(OUTPUT)
  }
  val narrowWidth = io.in.bits.data.getWidth / n
  require(io.in.bits.data.getWidth % narrowWidth == 0)

  if(n == 1) {
    io.in <> io.out
    io.cnt := UInt(width = 0)
    io.done := Bool(true)
  } else {
    val cnt = Reg(init=UInt(0, width = log2Up(n)))
    val wrap = cnt === UInt(n-1)
    val rbits = Reg(io.in.bits.clone)
    val active = Reg(init=Bool(false))

    val shifter = Vec.fill(n){Bits(width = narrowWidth)}
    (0 until n).foreach {
      i => shifter(i) := rbits.data((i+1)*narrowWidth-1,i*narrowWidth)
    }

    io.done := Bool(false)
    io.cnt := cnt
    io.in.ready := !active
    io.out.valid := active || io.in.valid
    io.out.bits := io.in.bits
    when(!active && io.in.valid) {
      when(io.in.bits.hasData()) {
        cnt := Mux(io.out.ready, UInt(1), UInt(0))
        rbits := io.in.bits
        active := Bool(true)
      }
      io.done := !io.in.bits.hasData()
    }
    when(active) {
      io.out.bits := rbits
      io.out.bits.data := shifter(cnt)
      when(io.out.ready) {
        cnt := cnt + UInt(1)
        when(wrap) {
          cnt := UInt(0)
          io.done := Bool(true)
          active := Bool(false)
        }
      }
    }
  }
}
```

The generated Verilog is:

```verilog
module FlowThroughSerializer(
  ...
);

  wire T1;
  wire[-1:0] T0;

`ifndef SYNTHESIS
// synthesis translate_off
  assign T0 = {0{$random}};
// synthesis translate_on
`endif
  assign io_done = 1'h1;
  assign io_cnt = T1;
  assign T1 = {1'h0, T0};
  assign io_out_bits_g_type = io_in_bits_g_type;
  assign io_out_bits_is_builtin_type = io_in_bits_is_builtin_type;
  assign io_out_bits_manager_xact_id = io_in_bits_manager_xact_id;
  assign io_out_bits_client_xact_id = io_in_bits_client_xact_id;
  assign io_out_bits_data = io_in_bits_data;
  assign io_out_bits_addr_beat = io_in_bits_addr_beat;
  assign io_out_valid = io_in_valid;
  assign io_in_ready = io_out_ready;
endmodule
```

Note the wire with negative width...

aswaterman, 2015-06-18:
Notably, it's not a negative-width wire but a zero-width wire.  These are
legal Chisel, but backend logic is supposed to filter them all out before
code generation.  Jim might know something about this.

ucbjrl, 2015-06-18:
Ben, could you provide details on the how to reproduce this?

ben-k, 2015-06-19:
If you check out stock reference chip, cd to vsim, and hit `make` you should see this result.

ucbjrl, 2015-07-21:
This should be fixed by pr #466

closed 2015-07-21 by ucbjrl