ucb-bar/chisel2-deprecated #448: Windows Tool Chain

sirinath, 2015-07-06:
Is it possible to release a easy to use MS Windows toolchain for Chisel.

da-steve101, 2015-08-13:
The main requirement i can see is the C compiler for the emulator.
If i wanted to run on windows i would start by getting sbt and scala working inside [cygwin](http://www.cygwin.com). See this [link](http://www.scala-sbt.org/0.13/tutorial/Manual-Installation.html)
I haven't tried but i think you could get it working without too much trouble.
Maybe have a go and make a pull request to document how you did it for other people?

kionix, 2015-08-13:
I can confirm that chisel under Cygwin works fine. I have also used
IntelliJ IDEA in the Cygwin setup. You just need to run it from the Cygwin
shell to have PATH and other environment variables set. You can then go on
and clone chisel-tutorial git repository, for example, within the IDE and
run c++ simulations there. Make sure you have g++, make, bash, git, and
other essential dev packages installed in the Cygwin setup. I don't recall
the exact cygwin dependency list and unfortunately don't have a windows set
up to try now. Of course you'd also need the Windows version of JDK
installed as well. I think it's a good idea to add this in the
documentations.

sirinath, 2015-08-13:
Good idea to add this to the documentation.