ucb-bar/chisel2-deprecated #489: Floating-point related inaccuracy in log2 functions

zhemao, 2015-08-10:
The log2Up and log2Ceil functions can sometimes give incorrect answers for large powers of two because of rounding errors from using the floating point log functions.

For instance, the implementation of log2Ceil is ceil(log(x)/log(2)). If you try log2Ceil(1 << 29), we expect to get 29. However, log(29)/log(2) in double-precision floating point is 29.000000000000004. So the answer given is actually 30.

I'm not sure what the best way of fixing this problem is. Something like

    if (isPow2(x)) log2Floor(x) else ceil(log(x)/log(2))

might work.

CuppoJava, 2015-08-10:
A loop-based integer implementation of log2 is probably the best solution
here.
  -Patrick

aswaterman, 2015-08-10:
Yeah, probably can just do BigInt(x).bitLength or something.  I'll make
sure this is fixed in Chisel3.

sdtwigg, 2015-08-10:
I already did the digging through the java docs on BigInt and associated
testing to verify those functions:

Unsigned values can be calculated via math.max(uvalue.bitLength, 1)
// max ensures bitLength of 0 is 1 not 0, omit if you want bitLength of 0
to be 0.
(from
https://github.com/sdtwigg/gama/blob/master/chiselfrontend/src/main/scala/IRUtil.scala#L12
)

Signed values can be calculated via (svalue.bitLength + 1)
(from
https://github.com/sdtwigg/gama/blob/master/chiselfrontend/src/main/scala/IRUtil.scala#L22
)

aswaterman, 2015-08-10:
That gives you sizeof(uvalue), not ceil(log2(uvalue)), which differs when uvalue is a power of 2.  This is what chisel3 is doing:

    object log2Up {
      def apply(in: Int): Int = 1 max BigInt(in-1).bitLength
    }
    
    object log2Down {
      def apply(in: Int): Int = log2Up(in) - (if (isPow2(in)) 0 else 1)
    }

closed 2020-05-13 by zhemao