ucb-bar/chisel2-deprecated #501: No otherwise with switch statement ?

Martoni, 2015-08-13:
In documentation its explained that we could use otherwise keyword for other case in switch statement:
```Scala
switch(idx) {
  is(v1) { u1 };
  is(v2) { u2 };
  otherwise { ud };
}
```
But I can't manage to use it, I have this error in scala compilation : 
```
[error] /usr/local/projects/ctbs/firmware/cmlkmgmt/seq_ctrl_reg.scala:55: not found: value otherwise
[error]       otherwise {
```

My code is : 
```scala
    switch(io.avlslave.address){
      is(CTRL_REG) {
        /* XXX */
        io.avlslave.readdata :=
          byteenablefilter(io.avlslave.byteenable, UInt(0, avldatasize))
      };
      is(FRAMECNT_REG) {
         io.avlslave.readdata :=
          byteenablefilter(io.avlslave.byteenable, framecount)
      };
      is(IMAGE_REG) {
        io.avlslave.readdata :=
          byteenablefilter(io.avlslave.byteenable, img_size)
      };
      otherwise {
        io.avlslave.readdata :=
          byteenablefilter(io.avlslave.byteenable, UInt(0, avldatasize))
      };
    }
```

I can't find a switch example with otherwise keyword in Chisel tutorial. Do you think it's a bug or I'm doing something wrong ?

Thanks

da-steve101, 2015-08-13:
There is no otherwise in a switch statement. Each 'is' is interpreted as a when block.
The way chisel works is that you are defining a graph of nodes sequentially.
For example:

    val io = new Bundle {
      val in   = UInt(INPUT, 4)
      val cond = Bool(INPUT)
      val out = UInt(OUTPUT, 4)
    }
    io.out := UInt(0) // Define a connection from the literal 0 to the output
    when ( cond ) { io.out := io.in } // Override the connection above when cond is true

This will have a 'default' value of 0 but pass through 'io.in' when the cond is true. Alternatively

    val io = new Bundle {
      val in   = UInt(INPUT, 4)
      val cond = Bool(INPUT)
      val out = UInt(OUTPUT, 4)
    }
    when ( cond ) { io.out := io.in } // Define io.out to be io.in when cond
    io.out := UInt(0) // Override the graph so that io.out is always 0

This overrides your graph so that the output is always 0
So essentially the answer to your question is just put the code in your otherwise before your switch statement ;)

Martoni, 2015-08-13:
Thanks for this quick and clear response. In fact the tutorial document I used is too old. The otherwise statement is no more in tutorial now.

closed 2015-08-13 by Martoni