ucb-bar/chisel2-deprecated #508: Clock name has argument is not set in verilog backend

Martoni, 2015-08-20:
When we use a Clock as an argument of a Module class like this : 
```scala
class SequenceurSimple(c: Clock = null, avlClk: Clock) extends Module(c) {
```
The Verilog backend can't find the name 'avlClk' and assign a T name : 
```verilog
module SequenceurSimple(input clk, input T181, input reset,
```
To solve this 'problem' I added this line in my module : 
```scala
  if (!avlClk.named) avlClk.setName("avlClk")
```
Do you think it's possible to assign the argument name directly ?

da-steve101, 2015-08-20:
This is maybe a post for chisel users: https://groups.google.com/forum/#!forum/chisel-users

Try having "val avlClk" rather than just avlClk in the arguments

da-steve101, 2015-08-20:
That didn't work for you? Worked for me

```scala
scala> import Chisel._                                                                                                                                                                         
import Chisel._                                                                                                                                                                                

scala> class SubMod( val myClk : Clock ) extends Module {
     | val io = new Bundle {
     | val in = UInt(INPUT, 4)
     | val out = UInt(OUTPUT, 4)
     | }
     | io.out := Reg(next=io.in, clock=myClk)
     | }
defined class SubMod

scala> class TopMod extends Module {
     | val io = new Bundle {
     | val in = UInt(INPUT, 4)
     | val out = UInt(OUTPUT, 4)
     | }
     | val sub = Module(new SubMod(new Clock))
     | sub.io <> io
     | }
defined class TopMod

scala> chiselMain(Array[String]("--v",
     | "--targetDir", "test-outputs"),
     | () => Module(new TopMod()))
```
Created the following verilog

```verilog
module _read__iw__iw__iw__iw_SubMod(input myClk,
    input [3:0] io_in,
    output[3:0] io_out
);

  reg [3:0] R0;

`ifndef SYNTHESIS
// synthesis translate_off
  integer initvar;
  initial begin
    #0.002;
    R0 = {1{$random}};
  end
// synthesis translate_on
`endif

  assign io_out = R0;

  always @(posedge myClk) begin
    R0 <= io_in;
  end
endmodule

module _read__iw__iw__iw__iw_TopMod(input myClk,
    input [3:0] io_in,
    output[3:0] io_out
);

  wire[3:0] sub_io_out;

  assign io_out = sub_io_out;
  _read__iw__iw__iw__iw_SubMod sub(.myClk(myClk),
       .io_in( io_in ),
       .io_out( sub_io_out )
  );
endmodule
```

Martoni, 2015-08-21:
No, that doesn't work. Full code is : 
```scala
import Chisel._
import AvlSlave._

class SeqCtrlReg(c: Clock = null, val serdesClk: Clock) extends Module(c) {
  val io = new Bundle {
    /* ddr_img_write input*/
    val img_size_i = Valid(UInt(INPUT, 19)).flip()
    val img_num_i = UInt(INPUT, 1)
    val coefftablenum_i = UInt(INPUT, 1)
    val framecount_i = Valid(UInt(INPUT, 32)).flip()
    val switch_coeff_flg_i = UInt(INPUT, 1)
    val switch_coeff_flg_o = UInt(OUTPUT, 1)
    val initframe = Bool(OUTPUT)

    /* ddr_img_write output*/
    val img_size_o = UInt(OUTPUT, 19)
    val img_num_o = Bool(OUTPUT)

    /* ddr coeff read */
    /* avalon-MM */
    val avlslave = new AvlSRW(addrWidth=2, dataWidth=32)
  }

  /* Clocks */
  /* Main avl clocks -> 125Mhz */
  /* serdesClk clocks -> < 50Mhz */
//  if(!serdesClk.named) serdesClk.setName("serdesClk")
  val serclkdbg = Reg(init=Bool(true), clock=serdesClk)
  serclkdbg := ~serclkdbg
  debug(serclkdbg)

  val avldatasize = io.avlslave.readdata.getWidth()
  val avladdrsize = io.avlslave.address.getWidth()

  def risingedge(x: Bool, inclock: Clock) = x && !Reg(next=x, clock=inclock)

  def byteenablefilter(byteenable: UInt, value: UInt) : UInt = {
    val byteenablesize = byteenable.getWidth()
    assert(Bool(byteenablesize == value.getWidth()/8),
           "Wrong byteenable, value size in byteenablefilter call")
    val bitmask = Cat((0 until byteenablesize).map {
                      i => Fill(8, byteenable(i))
                      // need to reverse otherwise endianness will be swapped
                      }.reverse)
    value & bitmask
  }

  def closedLoopSynchronize(inputsig: Bool,
                            inputclock: Clock,
                            outputclock: Clock) : Bool =
  {
    val outputsig = Reg(init=Bool(false), clock=inputclock)
    val output1 = Reg(init=Bool(false), clock=outputclock)
    val output2 = Reg(init=Bool(false), clock=outputclock)
    val ack1 = Reg(init=Bool(false), clock=inputclock)
    val ack2 = Reg(init=Bool(false), clock=inputclock)

    ack1 := output2
    ack2 := ack1

    output1 := outputsig
    output2 := output1

    when(inputsig) {
      outputsig := Bool(true)
    }.elsewhen((!inputsig) && ack2) {
      outputsig := Bool(false)
    }.otherwise {
[... 103 lines cut ...]
make -C /usr/local/projects/ctbs/firmware/cocotb/SeqCtrlReg/../../cmlkmgmt/ SeqCtrlReg.v
make[2]: Entering directory '/usr/local/projects/ctbs/firmware/cmlkmgmt'
echo "Converting chisel source to verilog"
Converting chisel source to verilog
sbt 'run-main SeqCtrlRegMain --backend v'
[info] Set current project to cmlkmgmt (in build file:/usr/local/projects/ctbs/firmware/cmlkmgmt/)
[info] Running SeqCtrlRegMain --backend v
[info] [0,110] // COMPILING < (class SeqCtrlReg)>(0)
[error] seq_ctrl_reg.scala:167: < Chisel.Clock@0 > doesn't have its component, yet. in class SeqCtrlRegMain$
Re-running Chisel in debug mode to obtain erroneous line numbers...
[info] [0,150] // COMPILING < (class SeqCtrlReg)>(0)
[error] seq_ctrl_reg.scala:167: < Chisel.Clock@0 > doesn't have its component, yet. in class SeqCtrlRegMain$
[error] (run-main-0) Chisel.ChiselException: < Chisel.Clock@0 > doesn't have its component, yet.
Chisel.ChiselException: < Chisel.Clock@0 > doesn't have its component, yet.
	at SeqCtrlRegMain$.main(seq_ctrl_reg.scala:167)
[trace] Stack trace suppressed: run last compile:runMain for the full output.
java.lang.RuntimeException: Nonzero exit code: 1
	at scala.sys.package$.error(package.scala:27)
[trace] Stack trace suppressed: run last compile:runMain for the full output.
[error] (compile:runMain) Nonzero exit code: 1
[error] Total time: 1 s, completed 21 août 2015 08:17:04
Makefile:64: recipe for target 'SeqCtrlReg.v' failed
make[2]: *** [SeqCtrlReg.v] Error 1
make[2]: Leaving directory '/usr/local/projects/ctbs/firmware/cmlkmgmt'
```

da-steve101, 2015-08-21:
So the issue there is that you are creating the clock outside of a Module() call.
You have to create the clock from inside your module.
ie) < Chisel.Clock@0 > doesn't have its component, yet. in class SeqCtrlRegMain$

Martoni, 2015-08-21:
I've got the same error with this line : 
```scala
    chiselMain(args, () => Module(new SeqCtrlReg(serdesClk=new Clock(reset=Driver.implicitReset))))
```

error : 
```bash
sbt 'run-main SeqCtrlRegMain --backend v'
[info] Set current project to cmlkmgmt (in build file:/usr/local/projects/ctbs/firmware/cmlkmgmt/)
[info] Compiling 1 Scala source to /usr/local/projects/ctbs/firmware/cmlkmgmt/target/scala-2.10/classes...
[info] Running SeqCtrlRegMain --backend v
[info] [0,125] // COMPILING < (class SeqCtrlReg)>(0)
[error] seq_ctrl_reg.scala:166: < Chisel.Clock@0 > doesn't have its component, yet. in class SeqCtrlRegMain$
Re-running Chisel in debug mode to obtain erroneous line numbers...
[info] [0,145] // COMPILING < (class SeqCtrlReg)>(0)
[error] seq_ctrl_reg.scala:166: < Chisel.Clock@449 > doesn't have its component, yet. in class SeqCtrlRegMain$
[error] (run-main-0) Chisel.ChiselException: < Chisel.Clock@449 > doesn't have its component, yet.
Chisel.ChiselException: < Chisel.Clock@449 > doesn't have its component, yet.
	at SeqCtrlRegMain$.main(seq_ctrl_reg.scala:166)
[trace] Stack trace suppressed: run last compile:runMain for the full output.
java.lang.RuntimeException: Nonzero exit code: 1
	at scala.sys.package$.error(package.scala:27)
[trace] Stack trace suppressed: run last compile:runMain for the full output.
[error] (compile:runMain) Nonzero exit code: 1
[error] Total time: 8 s, completed 21 août 2015 09:00:06
Makefile:64: recipe for target 'SeqCtrlReg.v' failed
```

> If not you need to move the creation of your clock inside the module

But if I do that, I can't pass the clock as an argument, no? And I have to do this to configure the clock frequency in the 'top' module.

da-steve101, 2015-08-21:
Can't you pass in the clock frequency as the argument instead?
I do think that clocks should be done differently, but creation within the module makes sense as the clock is still an input and output. Passing configuration such as frequency should be done as an argument

Martoni, 2015-08-23:
> Can't you pass in the clock frequency as the argument instead?

It could be a good solution only if clock used under this modules. But this clock is also used under other module in design. And all interfaces signals must be synchronous.

da-steve101, 2015-08-24:
Yeah sure, but aren't they going to be instantiated from a top level module still within a Module() call?
If not how are you running a simulation with multiple top level modules?
If your not and just generating verilog after simulating each one, you still need to instantiate all modules in a top level module in verilog. You can just attach the clocks there.

I needed to pass in a module to a another module for instantiation in that. The way I did it was to pass a function that returned a Module. You could do something similar:

```scala
var myClk = Option[Clock]
def getClock : Clock {
    myClk match {
        case None: => { tmpClk = new Clock; myClk = Some(tmpClk); tmpClk }
        case Some(tmpClk): => tmpClk
    }
}
...
// Pass it in to Module
class SeqCtrlReg(getClk : () => Clock) extends Module(c) {
val myClk = getClk()
...
```

closed 2016-03-09 by Martoni