ucb-bar/chisel2-deprecated #510: LargeNumber fail on master

andreaTP, 2015-08-20:
testing master before publishing it locally generate this error:
```
error]: Extract(4) out of range [0,3] of + [ ~(/*DataSuite_CarryChainComp_1__io_r in <DataSuite_CarryChainComp_1 (class DataSuite$CarryChainComp$1)>*/ Chisel.UInt(INPUT, width=4, connect to 0 inputs: ()))]+[  /*DataSuite_CarryChainComp_1__io_p in <DataSuite_CarryChainComp_1 (class DataSuite$CarryChainComp$1)>*/ Chisel.UInt(INPUT, width=4, connect to 0 inputs: ())]
Re-running Chisel in debug mode to obtain erroneous line numbers...
CPP elaborate
[info] [0.007] // COMPILING < (class DataSuite$CarryChainComp$1)>(0)
[info] [0.008] giving names
[info] [0.008] executing custom transforms
[info] [0.008] adding clocks and resets
[info] [0.008] inferring widths
[info] [0.008] checking widths
[info] [0.008] lowering complex nodes to primitives
[info] [0.008] removing type nodes
[info] [0.008] compiling 20 nodes
[info] [0.008] computing memory ports
[info] [0.008] resolving nodes to the components
[info] [0.009] creating clock domains
[info] [0.009] pruning unconnected IOs
[info] [0.009] checking for combinational loops
[info] [0.009] NO COMBINATIONAL LOOP FOUND
[info] [0.009] populating clock domains
[error] DataTest.scala:312: Extract(4) out of range [0,3] of + [ ~(/*DataSuite_CarryChainComp_1__io_r in <DataSuite_CarryChainComp_1 (class DataSuite$CarryChainComp$1)>*/ Chisel.UInt(INPUT, width=4, connect to 0 inputs: ()))]+[  /*DataSuite_CarryChainComp_1__io_p in <DataSuite_CarryChainComp_1 (class DataSuite$CarryChainComp$1)>*/ Chisel.UInt(INPUT, width=4, connect to 0 inputs: ())] in class DataSuite$CarryChainComp$1

widthInfinInfer ...
CPP elaborate
[info] [0.004] // COMPILING < (class DataSuite$BypassDataComp$1)>(0)
[info] [0.004] giving names
[error]: All IO's must be ports (dir set): <valid_2 (Chisel.Bool)> in <DataSuite_BypassDataComp_1 (class DataSuite$BypassDataComp$1)>
[error]: All IO's must be ports (dir set): <valid_1 (Chisel.Bool)> in <DataSuite_BypassDataComp_1 (class DataSuite$BypassDataComp$1)>
[error]: All IO's must be ports (dir set): <valid_0 (Chisel.Bool)> in <DataSuite_BypassDataComp_1 (class DataSuite$BypassDataComp$1)>
Re-running Chisel in debug mode to obtain erroneous line numbers...
CPP elaborate
[info] [0.009] // COMPILING < (class DataSuite$BypassDataComp$1)>(0)
[info] [0.009] giving names
[error] DataTest.scala:256: All IO's must be ports (dir set): <valid_2 (Chisel.Bool)> in <DataSuite_BypassDataComp_1 (class DataSuite$BypassDataComp$1)> in class DataSuite$BypassDataComp$1
[error] DataTest.scala:256: All IO's must be ports (dir set): <valid_1 (Chisel.Bool)> in <DataSuite_BypassDataComp_1 (class DataSuite$BypassDataComp$1)> in class DataSuite$BypassDataComp$1
[error] DataTest.scala:256: All IO's must be ports (dir set): <valid_0 (Chisel.Bool)> in <DataSuite_BypassDataComp_1 (class DataSuite$BypassDataComp$1)> in class DataSuite$BypassDataComp$1
[info] [0.004] // COMPILING < (class DataSuite$Parent$1)>(1)
[info] [0.004] giving names
[error] DataTest.scala:287: assigning to a non parent module's output port: /*? in < (class DataSuite$Child$1)>*/ Chisel.UInt(OUTPUT, width=8, connect to 1 inputs: (?[Chisel.UInt] in DataSuite$Child$1)) RHS: /*? in < (class DataSuite$Parent$1)>*/ Chisel.UInt(INPUT, width=8, connect to 0 inputs: ()) in class DataSuite$Parent$1
Re-running Chisel in debug mode to obtain erroneous line numbers...
[info] [0.002] // COMPILING < (class DataSuite$Parent$1)>(1)
[info] [0.003] giving names
[error] DataTest.scala:287: assigning to a non parent module's output port: /*? in < (class DataSuite$Child$1)>*/ Chisel.UInt(OUTPUT, width=8, connect to 1 inputs: (?[Chisel.UInt] in DataSuite$Child$1)) RHS: /*? in < (class DataSuite$Parent$1)>*/ Chisel.UInt(INPUT, width=8, connect to 0 inputs: ()) in class DataSuite$Parent$1
```

andreaTP, 2015-08-20:
a more relevant output is this one:
```
STARTING test-outputs//LargeNumberSuite_Rsh_1
RESET 5
java.io.IOException: Stream closed
	at java.lang.ProcessBuilder$NullOutputStream.write(ProcessBuilder.java:433)
	at java.io.OutputStream.write(OutputStream.java:116)
	at java.io.BufferedOutputStream.flushBuffer(BufferedOutputStream.java:82)
	at java.io.BufferedOutputStream.flush(BufferedOutputStream.java:140)
	at sun.nio.cs.StreamEncoder.implFlush(StreamEncoder.java:297)
	at sun.nio.cs.StreamEncoder.flush(StreamEncoder.java:141)
	at java.io.OutputStreamWriter.flush(OutputStreamWriter.java:229)
	at java.io.BufferedWriter.flush(BufferedWriter.java:254)
	at Chisel.Tester.Chisel$Tester$$writeln(Tester.scala:89)
	at Chisel.Tester.Chisel$Tester$$sendCmd(Tester.scala:116)
	at Chisel.Tester$$anonfun$reset$1.apply$mcVI$sp(Tester.scala:288)
	at scala.collection.immutable.Range.foreach$mVc$sp(Range.scala:166)
	at Chisel.Tester.reset(Tester.scala:287)
	at Chisel.Tester.<init>(Tester.scala:446)
	at LargeNumberSuite$RshTests$1.<init>(LargeNumber.scala:210)
	at LargeNumberSuite$$anonfun$testRsh$1.apply(LargeNumber.scala:222)
	at LargeNumberSuite$$anonfun$testRsh$1.apply(LargeNumber.scala:222)
	at Chisel.Driver$.apply(Driver.scala:53)
	at Chisel.chiselMain$.apply(hcl.scala:63)
	at Chisel.chiselMainTest$.apply(hcl.scala:76)
	at TestSuite.launchTester(TestSuite.scala:97)
	at TestSuite.launchCppTester(TestSuite.scala:105)
	at LargeNumberSuite.testRsh(LargeNumber.scala:222)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at org.junit.runners.model.FrameworkMethod$1.runReflectiveCall(FrameworkMethod.java:45)
	at org.junit.internal.runners.model.ReflectiveCallable.run(ReflectiveCallable.java:15)
	at org.junit.runners.model.FrameworkMethod.invokeExplosively(FrameworkMethod.java:42)
	at org.junit.internal.runners.statements.InvokeMethod.evaluate(InvokeMethod.java:20)
	at org.junit.internal.runners.statements.RunBefores.evaluate(RunBefores.java:28)
	at org.junit.runners.ParentRunner.runLeaf(ParentRunner.java:263)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:68)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:47)
	at org.junit.runners.ParentRunner$3.run(ParentRunner.java:231)
	at org.junit.runners.ParentRunner$1.schedule(ParentRunner.java:60)
	at org.junit.runners.ParentRunner.runChildren(ParentRunner.java:229)
	at org.junit.runners.ParentRunner.access$000(ParentRunner.java:50)
	at org.junit.runners.ParentRunner$2.evaluate(ParentRunner.java:222)
	at org.junit.runners.ParentRunner.run(ParentRunner.java:300)
	at org.junit.runners.Suite.runChild(Suite.java:128)
	at org.junit.runners.Suite.runChild(Suite.java:24)
	at org.junit.runners.ParentRunner$3.run(ParentRunner.java:231)
	at org.junit.runners.ParentRunner$1.schedule(ParentRunner.java:60)
	at org.junit.runners.ParentRunner.runChildren(ParentRunner.java:229)
	at org.junit.runners.ParentRunner.access$000(ParentRunner.java:50)
	at org.junit.runners.ParentRunner$2.evaluate(ParentRunner.java:222)
	at org.junit.runners.ParentRunner.run(ParentRunner.java:300)
	at org.junit.runner.JUnitCore.run(JUnitCore.java:157)
	at org.junit.runner.JUnitCore.run(JUnitCore.java:136)
	at org.junit.runner.JUnitCore.run(JUnitCore.java:117)
	at org.scalatest.junit.JUnitSuiteLike$class.run(JUnitSuiteLike.scala:260)
	at org.scalatest.junit.JUnitSuite.run(JUnitSuite.scala:79)
	at org.scalatest.tools.Framework.org$scalatest$tools$Framework$$runSuite(Framework.scala:466)
	at org.scalatest.tools.Framework$ScalaTestTask.execute(Framework.scala:677)
	at sbt.TestRunner.runTest$1(TestFramework.scala:76)
	at sbt.TestRunner.run(TestFramework.scala:85)
	at sbt.TestFramework$$anon$2$$anonfun$$init$$1$$anonfun$apply$8.apply(TestFramework.scala:202)
	at sbt.TestFramework$$anon$2$$anonfun$$init$$1$$anonfun$apply$8.apply(TestFramework.scala:202)
	at sbt.TestFramework$.sbt$TestFramework$$withContextLoader(TestFramework.scala:185)
	at sbt.TestFramework$$anon$2$$anonfun$$init$$1.apply(TestFramework.scala:202)
	at sbt.TestFramework$$anon$2$$anonfun$$init$$1.apply(TestFramework.scala:202)
	at sbt.TestFunction.apply(TestFramework.scala:207)
	at sbt.Tests$.sbt$Tests$$processRunnable$1(Tests.scala:239)
	at sbt.Tests$$anonfun$makeSerial$1.apply(Tests.scala:245)
	at sbt.Tests$$anonfun$makeSerial$1.apply(Tests.scala:245)
	at sbt.std.Transform$$anon$3$$anonfun$apply$2.apply(System.scala:44)
	at sbt.std.Transform$$anon$3$$anonfun$apply$2.apply(System.scala:44)
	at sbt.std.Transform$$anon$4.work(System.scala:63)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:226)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:226)
	at sbt.ErrorHandling$.wideConvert(ErrorHandling.scala:17)
	at sbt.Execute.work(Execute.scala:235)
	at sbt.Execute$$anonfun$submit$1.apply(Execute.scala:226)
	at sbt.Execute$$anonfun$submit$1.apply(Execute.scala:226)
	at sbt.ConcurrentRestrictions$$anon$4$$anonfun$1.apply(ConcurrentRestrictions.scala:159)
	at sbt.CompletionService$$anon$2.call(CompletionService.scala:28)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)
```
seems like the output stream is closed...

ucbjrl, 2015-08-20:
Some of the tests in the test suite are supposed to fail. This is the case with the testBuildCarryChain in DataTest.scala, so the exception is expected and if it isn't thrown, the test will fail.

On the other hand, all the LargeNumber tests are expected to pass. It might be the case that your c++ compilation step failed and there was no executable for the tester to run. We'll need to see more of the LargeNumber test output (specifically the 20 lines preceding the ones you've copied here) to be sure.

da-steve101, 2015-08-21:
Also please post your scala and g++ version

andreaTP, 2015-08-21:
These are complete std-out and std-err of the command ```sbt chisel/testOnly LargeNumberSuite```

https://gist.github.com/andreaTP/f8845c8c396cbb345550

https://gist.github.com/andreaTP/fcddbeea5407ab2a7327

scala is 2.11.6 as it is specified in build.sbt
g++ is
```Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 6.1.0 (clang-602.0.53) (based on LLVM 3.6.0svn)
Target: x86_64-apple-darwin14.5.0
Thread model: posix```

da-steve101, 2015-08-21:
Hmm strange, can you also post the generated cpp code? Should be in test-outputs/
called LargeNumberSuite_Rsh_1.cpp and LargeNumberSuite_Rsh_1.h

da-steve101, 2015-08-21:
Hmm ... Can't see any informative error messages.

Can you try running the following and let us know if it crashes:
```bash
g++ -g -c -o test-outputs//LargeNumberSuite_Rsh_1-emulator.o  -I../ -I/csrc/  test-outputs//LargeNumberSuite_Rsh_1-emulator.cpp
g++ -g -c -o test-outputs//LargeNumberSuite_Rsh_1.o  -I../ -I/csrc/  test-outputs//LargeNumberSuite_Rsh_1.cpp
g++ -g  -o test-outputs//LargeNumberSuite_Rsh_1 test-outputs//LargeNumberSuite_Rsh_1.o test-outputs//LargeNumberSuite_Rsh_1-emulator.o
gdb test-outputs/LargeNumberSuite_Rsh_1
```
Then run it. Hopefully it will tell us why it crashed

andreaTP, 2015-08-21:
I do not think this is useful:
```
sudo gdb -ex run test-outputs/LargeNumberSuite_Rsh_1
GNU gdb (GDB) 7.9.1
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-apple-darwin14.5.0".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from test-outputs/LargeNumberSuite_Rsh_1...done.
Starting program: /Users/andrea/Documents/workspace/chisel/test-outputs/LargeNumberSuite_Rsh_1
warning: `/BinaryCache/coreTLS/coreTLS-35.40.1~1/Objects/coretls.build/coretls.build/Objects-normal/x86_64/system_coretls_vers.o': can't open to read symbols: No such file or directory.
warning: Could not open OSO archive file "/BinaryCache/coreTLS/coreTLS-35.40.1~1/Symbols/BuiltProducts/libcoretls_ciphersuites.a"
warning: Could not open OSO archive file "/BinaryCache/coreTLS/coreTLS-35.40.1~1/Symbols/BuiltProducts/libcoretls_handshake.a"
warning: Could not open OSO archive file "/BinaryCache/coreTLS/coreTLS-35.40.1~1/Symbols/BuiltProducts/libcoretls_record.a"
warning: Could not open OSO archive file "/BinaryCache/coreTLS/coreTLS-35.40.1~1/Symbols/BuiltProducts/libcoretls_stream_parser.a"
600007fff5fbffb8000007fff5fc271b0
0000000010000000000007fff5fbffbb0
```

andreaTP, 2015-08-21:
this:
https://issues.scala-lang.org/browse/SI-7963
looks related to the problem

ucbjrl, 2015-08-21:
The stderr output is for a different test:

    ...

    at StdlibSuite.testOperators(StdlibTest.scala:165)
    ...

i.e., this one:

    [info] [0.014] // COMPILING < (class StdlibSuite$OperatorComp$1)>(0)
    [info] [0.016] giving names
    [error] StdlibTest.scala:73: UInt can't represent negative literal -1 in class StdlibSuite$OperatorComp$1
    [error] StdlibTest.scala:82: UInt can't represent negative literal -1 in class StdlibSuite$OperatorComp$1
    Re-running Chisel in debug mode to obtain erroneous line numbers...
    [info] [0.039] // COMPILING < (class StdlibSuite$OperatorComp$1)>(0)
    [info] [0.041] giving names
    [error] StdlibTest.scala:73: UInt can't represent negative literal -1 in class StdlibSuite$OperatorComp$1
    [error] StdlibTest.scala:82: UInt can't represent negative literal -1 in class StdlibSuite$OperatorComp$1
    java.lang.IllegalStateException: CODE HAS 2 ERRORS and 0 WARNINGS
        at Chisel.ChiselError$.checkpoint(ChiselError.scala:166)
        at Chisel.Backend.elaborate(Backend.scala:832)
        at Chisel.VerilogBackend.elaborate(Verilog.scala:690)
        at Chisel.Driver$.execute(Driver.scala:93)
        at Chisel.Driver$.apply(Driver.scala:41)
        at Chisel.Driver$.apply(Driver.scala:46)
        at Chisel.chiselMain$.apply(hcl.scala:60)
        at StdlibSuite.testOperators(StdlibTest.scala:165)

I think you need to include the test name with testOnly in quotes to actually have the intended effect:

    % sbt "chisel/testOnly LargeNumberSuite"

otherwise, it is silently ignored (and all tests are run).

da-steve101, 2015-08-24:
Replicated using clang++ on linux:

```
CPP elaborate
[info] [0.003] // COMPILING < (class LargeNumberSuite$Rsh$1)>(0)
[info] [0.003] giving names
[info] [0.004] executing custom transforms
[info] [0.004] adding clocks and resets
[info] [0.005] inferring widths
[info] [0.006] checking widths
[info] [0.006] lowering complex nodes to primitives
[info] [0.006] removing type nodes
[info] [0.007] compiling 8 nodes
[info] [0.007] computing memory ports
[info] [0.007] resolving nodes to the components
[info] [0.009] creating clock domains
[info] [0.010] pruning unconnected IOs
[info] [0.010] checking for combinational loops
[info] [0.010] NO COMBINATIONAL LOOP FOUND
[info] [0.015] populating clock domains
CppBackend::elaborate: need 0, redundant 0 shadow registers
[info] [0.018] generating cpp files
CppBackend: createCppFile LargeNumberSuite_Rsh_1.cpp
[info] [0.280] clang++ -c -o test-outputs//LargeNumberSuite_Rsh_1-emulator.o  -I../ -I/csrc/  test-outputs//LargeNumberSuite_Rsh_1-emulator.cpp RET 0
[info] [0.510] clang++ -c -o test-outputs//LargeNumberSuite_Rsh_1.o  -I../ -I/csrc/  test-outputs//LargeNumberSuite_Rsh_1.cpp RET 0
[info] [0.558] clang++   -o test-outputs//LargeNumberSuite_Rsh_1 test-outputs//LargeNumberSuite_Rsh_1.o test-outputs//LargeNumberSuite_Rsh_1-emulator.o RET 0
SEED 1440423390760
STARTING test-outputs//LargeNumberSuite_Rsh_1
RESET 5
CPP elaborate
[info] [0.002] // COMPILING < (class LargeNumberSuite$Rsh$2)>(0)
[info] [0.003] giving names
[info] [0.004] executing custom transforms
[info] [0.004] adding clocks and resets
[info] [0.004] inferring widths
[info] [0.005] checking widths
[info] [0.006] lowering complex nodes to primitives
[info] [0.006] removing type nodes
```

Can confirm no other exception or stderr output

da-steve101, 2015-08-24:
Running in gdb gives the following segmentation fault:

```
Program received signal SIGSEGV, Segmentation fault.
0x0000000000401f0f in rsha_n (d=0x7fffffffdf20, s0=0x7fffffffe1c0, amount=-136379611, nw=3, w=130) at test-outputs/emulator.h:216
216         d[i-n_shift_words] = val >> n_shift_bits | carry;
```

da-steve101, 2015-08-24:
Ok, just added a pull request that fixes it for clang
Something fishy is going on though, ill investigate further

da-steve101, 2015-08-24:
Resolved in pr #512

closed 2015-08-24 by ucbjrl

andreaTP, 2015-08-25:
I can confirm that now it works even on my Mac.
tnx guys!

PR #512 (2015-08-24): Add fix for rsh emulator which had seg fault on uninitialized values
Fixes #510 

Strange stuff is still happening
I don't know why but the for loop isn't terminating with clang

Work on it more later, don't merge yet