ucb-bar/chisel2-deprecated #513: Java stack error when a io signal not driven

Martoni, 2015-08-24:
I've got two signals declared in my io bundle : 
```scala
  val io = new Bundle {
    /* Outputs/inputs */
    val cc1 = Bool(OUTPUT)
[...]
    val trigger = Bool(INPUT)
  }
```

If I want to drive cc1  with when() like it: 
```scala
  when(io.trigger) {
    io.cc1 := Bool(true)
  }
```
I obtain an horrible java stack error : 
```java
[info] Set current project to cmlkmgmt (in build file:/usr/local/projects/ctbs/firmware/cmlkmgmt/)
[info] Compiling 1 Scala source to /usr/local/projects/ctbs/firmware/cmlkmgmt/target/scala-2.10/classes...
[info] Running CtrlCamMain --backend v
[info] [0,063] // COMPILING < (class CtrlCam)>(0)
[info] [0,073] giving names
[error] (run-main-0) java.lang.IllegalArgumentException: Flat hash tables cannot contain null elements.
java.lang.IllegalArgumentException: Flat hash tables cannot contain null elements.
	at scala.collection.mutable.FlatHashTable$HashUtils$class.elemHashCode(FlatHashTable.scala:390)
	at scala.collection.mutable.HashSet.elemHashCode(HashSet.scala:41)
	at scala.collection.mutable.FlatHashTable$class.findEntryImpl(FlatHashTable.scala:123)
	at scala.collection.mutable.FlatHashTable$class.containsEntry(FlatHashTable.scala:119)
	at scala.collection.mutable.HashSet.containsEntry(HashSet.scala:41)
	at scala.collection.mutable.HashSet.contains(HashSet.scala:58)
	at Chisel.Module.Chisel$Module$$walked$2(Module.scala:385)
	at Chisel.Module$$anonfun$pushInputs$1$1.apply(Module.scala:387)
	at Chisel.Module$$anonfun$pushInputs$1$1.apply(Module.scala:387)
	at scala.collection.TraversableLike$$anonfun$filterNot$1.apply(TraversableLike.scala:274)
	at scala.collection.TraversableLike$$anonfun$filterNot$1.apply(TraversableLike.scala:274)
	at scala.collection.TraversableLike$$anonfun$filter$1.apply(TraversableLike.scala:264)
	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:47)
	at scala.collection.TraversableLike$class.filter(TraversableLike.scala:263)
	at scala.collection.AbstractTraversable.filter(Traversable.scala:105)
	at scala.collection.TraversableLike$class.filterNot(TraversableLike.scala:274)
	at scala.collection.AbstractTraversable.filterNot(Traversable.scala:105)
	at Chisel.Module.pushInputs$1(Module.scala:387)
	at Chisel.Module.dfs(Module.scala:395)
	at Chisel.Backend$$anonfun$nameAll$1.apply(Backend.scala:205)
	at Chisel.Backend$$anonfun$nameAll$1.apply(Backend.scala:195)
	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:47)
	at Chisel.Backend.nameAll(Backend.scala:195)
	at Chisel.Backend.elaborate(Backend.scala:831)
	at Chisel.VerilogBackend.elaborate(Verilog.scala:690)
	at Chisel.Driver$.execute(Driver.scala:92)
	at Chisel.Driver$.apply(Driver.scala:40)
	at Chisel.chiselMain$.apply(hcl.scala:60)
	at CtrlCamMain$.main(ctrl_cam.scala:58)
	at CtrlCamMain.main(ctrl_cam.scala)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:57)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:606)
[trace] Stack trace suppressed: run last compile:runMain for the full output.
java.lang.RuntimeException: Nonzero exit code: 1
	at scala.sys.package$.error(package.scala:27)
[trace] Stack trace suppressed: run last compile:runMain for the full output.
[error] (compile:runMain) Nonzero exit code: 1
[error] Total time: 5 s, completed 24 août 2015 17:48:12
Makefile:74: recipe for target 'CtrlCam.v' failed
```
Which is not very clear. I think the error is that 'io.cc1' is not driven by default, if I fix the code to add a default value like this : 
```scala
  io.cc1 := Bool(false)
  when(io.trigger) {
    io.cc1 := Bool(true)
  }
```
That's ok.
Do you thinks it's possible to improve Chisel error message to have somethings more relevant than : 
```
Flat hash tables cannot contain null elements.
```
?

ucbjrl, 2015-08-24:
Should be fixed with commit 1b5aadd.

closed 2015-08-24 by ucbjrl

Martoni, 2015-08-25:
Thanks for this so quick fix !
I merged it with my local view of Chisel and it's worked, but note that this working only with scala version 2.11.1. With scala version 2.10.4 that is recommended in README.md the java stack error is still here.