ucb-bar/chisel2-deprecated #514: separate functionalities and DSL in packages under ```Chisel```

andreaTP, 2015-08-25:
I find quite hard to navigate the sources of Chisel at the moment.
What do you think about moving semantically different files under specific packages?

You could think that ```UInt``` , ```Bundle``` etc will go into a sub-package called ```Chisel.DSL```
And on the other hand ```Tester``` could be under ```Chisel.test```

I such way one can choose to integrate separate parts of the project and separate properly it's own application.

If you like this proposal but do not have time I can start with a proposal on that and we can discuss on it.

da-steve101, 2015-08-25:
I don't have any say over this but I somewhat agree. I don't think UInt/Bundle, but definitely the test stuff I reckon and Util stuff (Complex + ChiselUtils).
Issue with a change like that is that it breaks everything.
I don't think its bad enough to warrant breaking that.
As for getting lost in the code, i think whats needed is developer documentation. Just a brief high level description of how everything fits together, known issues and general direction of development

andreaTP, 2015-08-25:
I totally agree with the developer documentation stuff, although it looks like the project is young enough to make an early breaking change that will ease the future development (and that affect just the import statements).
At least you can think about a modularisation with separate sub-projects (with a multi project build) that can save package compatibility but separate responsibilities of code, and deploy them as an aggregate.

da-steve101, 2015-08-25:
A separate chisel utils project looks like its coming (https://groups.google.com/forum/#!topic/chisel-users/pxRCGfFL_YQ)
Perhaps Chisel 3 could break some compatibility and Chisel 2 remains supported with bug fixes. Not my decision though.