ucb-bar/chisel2-deprecated #515: Enable testing variable size modules

andreaTP, 2015-08-25:
The problem to solve here is that if I've developed a module that depends on an input parameter and I want to test it against multiple inputs.
an example could be this  https://github.com/ucb-bar/chisel-tutorial/blob/master/problems/problems.scala#L35

how to test something like
```for (i <- 2 to 10) yield Module(new Adder(i))```
?

unfortunately the signature of ```chiseMain``` is ```[T <: Module]``` so I can test one predefined module at time and I cannot have a factory of tests.

kionix, 2015-08-25:
You can easily do that:

    val mods = for (i <- 2 to 10) yield Module(new Adder(i))
    mods.foreach(mod => chiselMain(args, ()=> mod ) )

or maybe generate modules on the go:

    (2 to 10).foreach( i => chiselMain.run(args , () => new Adder(i)) )

andreaTP, 2015-08-26:
this drive me to spread around test related code, but if this is the only way to go ok thanks.