ucb-bar/chisel2-deprecated #521: Clock given in Module(c) argument is not always the Driver.implicitClock

Martoni, 2015-09-02:
I extended module in that way : 
```scala
class SeqCtrlReg(c: Clock = null, serdesClk: Clock) extends Module(c) {
}
```
I would have thought that the argument 'c' given in Module was the 'Driver.ImplicitClock'
But in fact, no.

If I instantiate a register like it under the SeqCtrlReg module : 
```scala
    val outputsig = Reg(init=Bool(false), clock=Driver.implicitClock)
```

And I instantiate the module in my 'top' like it : 
``` scala
  val seqctrlreg = Module(new SeqCtrlReg(c=avlClk, serdesClk=Driver.implicitClock))
``` 

The actual clock for outputsig will be serdesClk.

In fact I understand Driver is a singleton, then implicitClock is the same is all design submodules. But it really strange behavior no ?

da-steve101, 2015-09-03:
I think you may be confusing Driver.implicitClock and Module.clock
There is one implicit generated clock which is the default if no others are specified. This single clock is accessible everywhere using Driver.implicitClock 
However if you have
```scala
class SeqCtrlReg(c: Clock = null) extends Module(c) {
    val io = new Bundle {
      val in = UInt(INPUT, 4)
      val out = UInt(OUTPUT, 4)
    }
    Predef.assert(this.clock == c)
    io.out := Reg(next=io.in, clock=clock)
}
class TestTop extends Module {
    val io = new Bundle {
      val in = UInt(INPUT, 4)
      val out = UInt(OUTPUT, 4)
    }
    val myClk = new Clock()
    val seqinst = Module(new SeqCtrlReg(myClk))
    io <> seqinst.io
}
```
which works as expected for me.

Martoni, 2015-09-03:
Thanks da-steve101, I better understand it now.

closed 2015-09-03 by Martoni