ucb-bar/chisel2-deprecated #530: JVM stack overflow on illegal use of wire

wasabiz, 2015-09-13:
I wrote the following (illegal) code because I didn't figure out how to use wires and found that this causes java.lang.StackOverflowError. It should be better raising a saner error by
 checking if wires are properly declared and used.

```scala
class Foo extends Module {

  val io = new Bundle {
    val a = UInt(OUTPUT, width = 32)
  }

  val b = Reg(UInt(width = 32))

  val c = Reg(UInt(width = 32))

  val d = UInt(0, width = 32)

  d := b

  c := d

  io.a := c
}
```

da-steve101, 2015-09-14:
@ucbjrl it appears width_ = None for node d
Should assigning to a literal be disallowed (I think so) or should the width be assigned?
Perhaps both? Im surprised it isn't already defined if declared with UInt( width = ...), should it be?

ucbjrl, 2015-10-13:
Closed by #557 at  https://github.com/ucb-bar/chisel/commit/201c2ee9faefcdc79385dd393bef7c67abeadd1b

closed 2015-10-13 by ucbjrl