ucb-bar/chisel2-deprecated #549: Support for asynchronous resets in Verilog backend

jwright6323, 2015-09-26:
It appears that Chisel does not support asynchronous resets. Can you add an option to use asynchronous resets in generated Verilog? This would help with our pin-limited designs (among others). An option to select the polarity of reset would also be useful, as most libraries only contain flops with active-low async reset/set pins.

Thanks,
John

mciepluc, 2016-01-23:
Hello,

Just to add a comment to this feature request and a discussion from https://groups.google.com/forum/#!topic/chisel-users/4cc4SyB5mk8

Async resets with selected polarity would be more than welcome. However, please note it's possible that not all registers in the design are resetable using async reset. Some registers can be cleared using control signal and some of them may not have any reset at all. 
As Chisel by default generates a single always block with FFs with only a clk at the sensitivity list, it will be required to create 2 always blocks when async resets are implemented. 

E.g. 
```verilog
always @(posedge clk or negedge rst_n)
begin
if (~rst_n)
  rega <= 1'b0
  ... all async resetable flops
end

always @(posedge clk)
begin
if (rst)
  regb <= 1'b0
  ... all other flops
```
end

Please also note that the "if (~rst_n)" condition have to be the very first in the process and cannot contain any logic operation (synthesis tools requirement).

MC

mcieplucha, 2016-02-08:
Hello,

Please find some comments how asynchronous resets in generated code should be implemented, to make sure that ASIC synthesis rules are not violated.

1. In the design resettable and non-resettable flops may exist together. It is important that these flops are described in separated ```always``` blocks.
2. The asynchronous reset condition must be the very first in the ```always``` block and it must constain only reset signal (e.g. ```if (rst)``` or ```if (~rst_n)```).
3. Asynchronous reset signal should not be used for any logic outside ```always``` block (e.g. ```assign a = (rst) ? 1'b0 : b;``` is prohibited).
4. Reset polarity selection option is welcome, please note it is highly unlikely that both polarities coexist in single design. 

Below is an example how generated code should look like:
```verilog
always @(posedge clk or negedge rst_n)
begin
  if (~rst_n) begin //async-resettable flops
    flop1 <= FLOP1_RESET_VALUE;
    flop2 <= FLOP2_RESET_VALUE;
    //all async resettable flops
  end else //other conditions
end

always @(posedge clk)
begin
  if (rst) begin //sync-resettable flops (rst may be a local signal or any condition)
    flopA <= FLOPA_RESET_VALUE;
    flopB <= FLOPB_RESET_VALUE;
    //all other flops
  end else //other conditions
    //...
  flopC <= flopC_next; //this flop has no sync or async reset
end
```

Regards,
Marek Ciepłucha

jwright6323, 2019-08-13:
🎉freechipsproject/chisel3#1011 I guess I can close this now? Thanks @jackkoenig !

closed 2019-08-13 by jwright6323