ucb-bar/chisel2-deprecated #567: Chisel causing strange internal Scala compiler error

zhemao, 2015-10-22:
I tried pulling the latest master and now I'm getting a weird compiler error. Anybody know what's going on

    java.lang.AssertionError: assertion failed: 
      class Reg
         while compiling: /scratch/zhemao/rocket-chip/chisel/src/main/scala/Backend.scala
            during phase: erasure
         library version: version 2.11.6
        compiler version: version 2.11.6
      reconstructed args: -bootclasspath /usr/lib/jvm/java-7-openjdk-amd64/jre/lib/resources.jar:/usr/lib/jvm/java-7-openjdk-amd64/jre/lib/rt.jar:/usr/lib/jvm/java-7-openjdk-amd64/jre/lib/sunrsasign.jar:/usr/lib/jvm/java-7-openjdk-amd64/jre/lib/jsse.jar:/usr/lib/jvm/java-7-openjdk-amd64/jre/lib/jce.jar:/usr/lib/jvm/java-7-openjdk-amd64/jre/lib/charsets.jar:/usr/lib/jvm/java-7-openjdk-amd64/jre/lib/rhino.jar:/usr/lib/jvm/java-7-openjdk-amd64/jre/lib/jfr.jar:/usr/lib/jvm/java-7-openjdk-amd64/jre/classes:/home/eecs/zhemao/.ivy2/cache/org.scala-lang/scala-library/jars/scala-library-2.11.6.jar -deprecation -feature -classpath /scratch/zhemao/rocket-chip/chisel/target/scala-2.11/classes:/home/eecs/zhemao/.ivy2/cache/org.scala-lang/scala-reflect/jars/scala-reflect-2.11.6.jar -language:reflectiveCalls -language:implicitConversions -language:existentials -unchecked
    
      last tree to typer: TypeTree(trait SeqLike)
           tree position: line 200 of /scratch/zhemao/rocket-chip/chisel/src/main/scala/Backend.scala
                tree tpe: collection.SeqLike
                  symbol: abstract trait SeqLike in package collection
       symbol definition: abstract trait SeqLike[+A, +Repr] extends Any with IterableLike[A,Repr] with GenSeqLike[A,Repr] with Parallelizable[A,scala.collection.parallel.ParSeq[A]] (a ClassSymbol)
          symbol package: scala.collection
           symbol owners: trait SeqLike
               call site: method apply in package Chisel
    
    == Source file context for tree position ==
    
       197         // ensure this component has a name
       198       val children = childrenOfParent.getOrElse(comp, Seq.empty)
       199       assert(children.filter(_.name.isEmpty).isEmpty, ChiselError.error("Internal Error: Unnamed Children"))
       200         // since sortedComps, all children should have names due to check above
       201       
       202       // ensure all nodes in the design has SOME name
       203       comp dfs { 
    	at scala.tools.nsc.transform.AddInterfaces.transformMixinInfo(AddInterfaces.scala:217)
    	at scala.tools.nsc.transform.Erasure.transformInfo(Erasure.scala:375)
    	at scala.tools.nsc.transform.InfoTransform$Phase$$anon$1.transform(InfoTransform.scala:38)
    	at scala.reflect.internal.Symbols$Symbol.rawInfo(Symbols.scala:1578)
    	at scala.reflect.internal.Symbols$Symbol.info(Symbols.scala:1498)
    	at scala.reflect.internal.Kinds$$anonfun$checkKindBounds0$2.apply(Kinds.scala:215)
    	at scala.reflect.internal.Kinds$$anonfun$checkKindBounds0$2.apply(Kinds.scala:211)
    	at scala.reflect.internal.util.Collections$class.flatMap2(Collections.scala:130)
    	at scala.reflect.internal.SymbolTable.flatMap2(SymbolTable.scala:16)
    	at scala.reflect.internal.Kinds$class.checkKindBounds0(Kinds.scala:211)
    	at scala.reflect.internal.SymbolTable.checkKindBounds0(SymbolTable.scala:16)
    	at scala.tools.nsc.typechecker.Infer$Inferencer.checkKindBounds(Infer.scala:902)
    	at scala.tools.nsc.typechecker.Infer$Inferencer.check$1(Infer.scala:893)
    	at scala.tools.nsc.typechecker.Infer$Inferencer.checkBounds(Infer.scala:898)
    [error] (chisel/compile:compileIncremental) java.lang.AssertionError: assertion failed: 
    [error]   class Reg
    [error]      while compiling: /scratch/zhemao/rocket-chip/chisel/src/main/scala/Backend.scala
    [error]         during phase: erasure
    [error]      library version: version 2.11.6
    [error]     compiler version: version 2.11.6
    [error]   reconstructed args: -bootclasspath /usr/lib/jvm/java-7-openjdk-amd64/jre/lib/resources.jar:/usr/lib/jvm/java-7-openjdk-amd64/jre/lib/rt.jar:/usr/lib/jvm/java-7-openjdk-amd64/jre/lib/sunrsasign.jar:/usr/lib/jvm/java-7-openjdk-amd64/jre/lib/jsse.jar:/usr/lib/jvm/java-7-openjdk-amd64/jre/lib/jce.jar:/usr/lib/jvm/java-7-openjdk-amd64/jre/lib/charsets.jar:/usr/lib/jvm/java-7-openjdk-amd64/jre/lib/rhino.jar:/usr/lib/jvm/java-7-openjdk-amd64/jre/lib/jfr.jar:/usr/lib/jvm/java-7-openjdk-amd64/jre/classes:/home/eecs/zhemao/.ivy2/cache/org.scala-lang/scala-library/jars/scala-library-2.11.6.jar -deprecation -feature -classpath /scratch/zhemao/rocket-chip/chisel/target/scala-2.11/classes:/home/eecs/zhemao/.ivy2/cache/org.scala-lang/scala-reflect/jars/scala-reflect-2.11.6.jar -language:reflectiveCalls -language:implicitConversions -language:existentials -unchecked
    [error] 
    [error]   last tree to typer: TypeTree(trait SeqLike)
    [error]        tree position: line 200 of /scratch/zhemao/rocket-chip/chisel/src/main/scala/Backend.scala
    [error]             tree tpe: collection.SeqLike
    [error]               symbol: abstract trait SeqLike in package collection
    [error]    symbol definition: abstract trait SeqLike[+A, +Repr] extends Any with IterableLike[A,Repr] with GenSeqLike[A,Repr] with Parallelizable[A,scala.collection.parallel.ParSeq[A]] (a ClassSymbol)
    [error]       symbol package: scala.collection
    [error]        symbol owners: trait SeqLike
    [error]            call site: method apply in package Chisel
    [error] 
    [error] == Source file context for tree position ==
    [error] 
    [error]    197         // ensure this component has a name
    [error]    198       val children = childrenOfParent.getOrElse(comp, Seq.empty)
    [error]    199       assert(children.filter(_.name.isEmpty).isEmpty, ChiselError.error("Internal Error: Unnamed Children"))
    [error]    200         // since sortedComps, all children should have names due to check above
    [error]    201       
    [error]    202       // ensure all nodes in the design has SOME name
    [error]    203       comp dfs { 
    [error] Total time: 25 s, completed Oct 21, 2015 6:38:16 PM

donggyukim, 2015-10-22:
I don't know why this happens, but run <code>sbt clean</code> in the chisel directory, and run your stuff again.

colinschmidt, 2015-10-22:
Palmer says bisect it and post the commit.

zhemao, 2015-10-22:
Turns out this is because some intermediate build output was screwing up sbt. Deleting all my target directories and retrying solved the issue.

closed 2015-10-22 by zhemao