ucb-bar/chisel2-deprecated #569: Combinational Loop Detector Is Broken

wasabiz, 2015-10-31:
I just encountered a strange bug after updated chisel to the latest release. The following code does not compile as chisel complains there is a combinational loop found, which there apparently ain't. 

```scala
import Chisel._

class SdramController extends Module {

  val io = new Bundle {
    val a = Bits(OUTPUT, 13)
  }

  io.a := UInt(0)
  io.a(10) := Bool(true)
}

object SdramController {
  def main(args: Array[String]) {
    chiselMainTest (args, () => Module(new SdramController())) { c => new Tester(c){} }
  }
}
```

A part of compile log follows:

```
Re-running Chisel in debug mode to obtain erroneous line numbers...                                                                                                                  
CPP elaborate                                                                                                                                                                        
[info] [0.013] // COMPILING < (class SdramController)>(0)                                                                                                                            
[info] [0.015] giving names                                                                                                                                                          
[info] [0.016] executing custom transforms                                                                                                                                           
[info] [0.016] adding clocks and resets                                                                                                                                              
[info] [0.016] inferring widths                                                                                                                                                      
[info] [0.017] checking widths                                                                                                                                                       
[info] [0.018] lowering complex nodes to primitives                                                                                                                                  
[info] [0.019] removing type nodes                                                                                                                                                   
[info] [0.019] compiling 15 nodes                                                                                                                                                    
[info] [0.020] computing memory ports                                                                                                                                                
[info] [0.020] resolving nodes to the components                                                                                                                                     
[info] [0.048] creating clock domains                                                                                                                                                
[info] [0.048] pruning unconnected IOs                                                                                                                                               
[info] [0.048] checking for combinational loops                                                                                                                                      
[error] SdramController.scala:15: FOUND COMBINATIONAL PATH! in class SdramController$                                                                                                
[error] SdramController.scala:10:   (0) in class SdramController                                                                                                                     
[error] SdramController.scala:10:   (1) in class SdramController                                                                                                                     
[error] SdramController.scala:10:   (2) in class SdramController                                                                                                                     
[error] SdramController.scala:10:   (3) in class SdramController                                                                                                                     
[error] SdramController.scala:10:   (4) in class SdramController                                                                                                                     
[error] SdramController.scala:10:   (5) in class SdramController                                                                                                                     
[error] SdramController.scala:6:   (6) in class SdramController$$anon$1                                                                                                              
[error] (run-main-0) java.lang.IllegalStateException: CODE HAS 8 ERRORS and 0 WARNINGS                                                                                               
java.lang.IllegalStateException: CODE HAS 8 ERRORS and 0 WARNINGS                                                                                                                    
        at Chisel.ChiselError$.checkpoint(ChiselError.scala:158)                                                                                                                     
        at Chisel.Backend.elaborate(Backend.scala:906)                                                                                                                               
        at Chisel.CppBackend.elaborate(Cpp.scala:1360)                                                                                                                               
        at Chisel.Driver$.execute(Driver.scala:93)                                                                                                                                   
        at Chisel.Driver$.apply(Driver.scala:41)                                                                                                                                     
        at Chisel.Driver$.apply(Driver.scala:46)                                                                                                                                     
        at Chisel.Driver$.apply(Driver.scala:52)                                                                                                                                     
        at Chisel.chiselMain$.apply(hcl.scala:63)                                                                                                                                    
        at Chisel.chiselMainTest$.apply(hcl.scala:76)                                                                                                                                
        at SdramController$.main(SdramController.scala:15)                                                                                                                           
        at SdramController.main(SdramController.scala)                                                                                                                               
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)                                                                                                               
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)                                                                                             
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.j                                                                                            
ava:43)                                                                                                                                                                              
        at java.lang.reflect.Method.invoke(Method.java:497)                                                                                                                          
[trace] Stack trace suppressed: run last compile:run for the full output.
```

donggyukim, 2015-11-01:
Internally, a combination loop is constructed with a subword assignment to a wire(<code>io.a(10) := Bool(true)</code>) The previous combinational loop checker was broken and didn't catch it. I think we should fix it, but for now, you can implement it differently using shifts or turn it off with the --noCombLoop flag.

wasabiz, 2015-11-02:
Thanks for the reply. I'll use noCombLoop option for a moment.

commit 971786f6 (2015-12-01): break comb loops in subwords

should fix #569
closed 2015-12-02 by donggyukim