ucb-bar/chisel2-deprecated #583: next on empty iterator

ascenium, 2015-11-10:
It seems that when I instantiate a fair number of sizable modules at once, I hit this error as I go into the C++ simulation.  Here is the entire run listing:

BigKiss:chisel mykland$ sbt run
[info] Set current project to chisel (in build file:/Users/mykland/work/chisel/)
[info] Compiling 2 Scala sources to /Users/mykland/work/chisel/target/scala-2.11/classes...
[info] Running ascenium.mainStub 
CPP elaborate
[info] [1.401] // COMPILING < (class ascenium.array)>(128)
[info] [2.973] giving names
[info] [4.004] executing custom transforms
[info] [4.004] adding clocks and resets
[info] [4.580] inferring widths
[info] [5.696] checking widths
[info] [6.025] lowering complex nodes to primitives
[info] [6.025] removing type nodes
[info] [6.435] compiling 196611 nodes
[info] [6.435] computing memory ports
[info] [7.065] resolving nodes to the components
[info] [9.349] creating clock domains
[info] [9.357] pruning unconnected IOs
[info] [9.464] checking for combinational loops
[info] [10.253] NO COMBINATIONAL LOOP FOUND
[info] [10.962] populating clock domains
CppBackend::elaborate: need 0, redundant 0 shadow registers
[info] [12.877] generating cpp files
CppBackend: createCppFile array.cpp
[info] [24.073] g++ -c -o ./array-emulator.o  -I../ -I/csrc/  ./array-emulator.cpp RET 0
[info] [228.935] g++ -c -o ./array.o  -I../ -I/csrc/  ./array.cpp RET 0
[info] [235.684] g++   -o ./array ./array.o ./array-emulator.o RET 0
SEED 1447197098452
STARTING ./array
[error] (run-main-0) java.util.NoSuchElementException: next on empty iterator
java.util.NoSuchElementException: next on empty iterator
	at scala.collection.Iterator$$anon$2.next(Iterator.scala:39)
	at scala.collection.Iterator$$anon$2.next(Iterator.scala:37)
	at scala.collection.IndexedSeqLike$Elements.next(IndexedSeqLike.scala:63)
	at scala.collection.IterableLike$class.head(IterableLike.scala:107)
	at scala.collection.mutable.ArrayBuffer.scala$collection$IndexedSeqOptimized$$super$head(ArrayBuffer.scala:48)
	at scala.collection.IndexedSeqOptimized$class.head(IndexedSeqOptimized.scala:126)
	at scala.collection.mutable.ArrayBuffer.head(ArrayBuffer.scala:48)
	at scala.collection.TraversableLike$class.last(TraversableLike.scala:459)
	at scala.collection.mutable.ArrayBuffer.scala$collection$IndexedSeqOptimized$$super$last(ArrayBuffer.scala:48)
	at scala.collection.IndexedSeqOptimized$class.last(IndexedSeqOptimized.scala:132)
	at scala.collection.mutable.ArrayBuffer.last(ArrayBuffer.scala:48)
	at Chisel.Tester.Chisel$Tester$$mwhile(Tester.scala:117)
	at Chisel.Tester.start(Tester.scala:622)
	at Chisel.Tester.<init>(Tester.scala:652)
	at ascenium.array_Tests.<init>(test.scala:36)
	at ascenium.mainStub$$anonfun$main$2.apply(test.scala:49)
	at ascenium.mainStub$$anonfun$main$2.apply(test.scala:49)
	at Chisel.Driver$.apply(Driver.scala:53)
	at Chisel.chiselMain$.apply(hcl.scala:63)
	at Chisel.chiselMainTest$.apply(hcl.scala:76)
	at ascenium.mainStub$.main(test.scala:48)
	at ascenium.mainStub.main(test.scala)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
[trace] Stack trace suppressed: run last compile:run for the full output.
java.lang.RuntimeException: Nonzero exit code: 1
	at scala.sys.package$.error(package.scala:27)
[trace] Stack trace suppressed: run last compile:run for the full output.
[error] (compile:run) Nonzero exit code: 1
[error] Total time: 261 s, completed Nov 10, 2015 3:15:52 PM

How rude, I can't drag and drop the scala source file.  But here it is:

package ascenium

import Chisel._

class test extends Module
{
	val io = new Bundle
	{
		val in0s = Vec( 256, UInt( INPUT, 16 ) )
		val in1s = Vec( 256, UInt( INPUT, 16 ) )
		val outs = Vec( 256, UInt( OUTPUT, 32 ) )
	}
	for( i <- 0 to 255 )
	{
		io.outs(i) := io.in0s(i) * io.in0s(i) + io.in1s(i) * io.in1s(i)
	}
}

class array extends Module
{
	val io = new Bundle
	{
		val in0 = UInt( INPUT, 16 )
		val in1 = UInt( INPUT, 16 )
		val out = UInt( OUTPUT, 32 )
    }

	// Modules
	// Note that we can't create a vector of modules directly.  Vectors only apply
	// to wires or registers.  However, we can create the modules inside the
	// statement and fill a vector with their wires, thereby achieving our goal
	// indirectly.
	val elements	= Vec.fill( 128 ) { Module( new test ).io }
}

class array_Tests(c: array) extends Tester(c)
{
	step( 1 )
}

object mainStub
{
	def main( args: Array[String] ): Unit =
	{
    	chiselMainTest( Array[String]("--backend", "c", // "--backend", "v",
    			"--compile", "--test", "--genHarness"),
    			() => Module( new array() ) )
    	{
    		c => new array_Tests( c )
    	}
	}
}

Note that in my actual code, the module test is instead a fairly complex module that does a bunch of interesting stuff.  When I do a Vec.fill() of 64 instead of 128, it works fine.  If I do two Vec.fill() of 32, it works fine.

ascenium, 2015-11-11:
More information on this bug.  This is undoubtedly a resource related issue.  On my laptop having more limited resources, the issue is repeatable.  On my desktop machine, again, this same code that
fails on my laptop runs fine.  In stressing the code in related ways, I have triggered Java garbage collection and stack space failures.  Is there a recommendation for these Java parameters for Chisel that should be documented?

da-steve101, 2015-11-12:
I suspect that the binary generated 'array' is crashing possibly due to memory issues. Taking a stab in the dark but i think the issue is 'array' is crashing hence the scala/java can't communicate with it. You can run the binary by itself to get more information. 'array' is the C binary so messing with java wouldn't cause it to replicate (if thats the problem). Compare the c compilers on your laptop and desktop too

ucbjrl, 2015-11-24:
The 'array' test application is exiting early with a SIGSEGV before producing any output. PR #591 should address this.

closed 2015-11-24 by ucbjrl