ucb-bar/chisel2-deprecated #623: Peeking INPUT before Poking

shunshou, 2015-12-15:
What's the expected behavior when you peek an input before poking it? Before, peeking would just return 0x000... (I think? It's been a while since I peeked before poked... lol), but now, I get a java.util.NoSuchElementException: key not found error. 

I assume it's b/c this line of peek:

else if (data.isTopLevelIO && data.dir == INPUT) _pokeMap(data)

expects _pokeMap(data) to contain something when it doesn't if you haven't poked the node yet. 

I'm not objecting to having the tester error out, but a more descriptive error message would be nice...

donggyukim, 2016-01-08:
I didn't expect this situation. Would it be ok to return 0?

shunshou, 2016-01-11:
I'm personally ok with it returning 0, but maybe others will object? Since that assumes input nodes have a default value of 0 without it being explicitly specified? Alternatively, rather than saying key not found, you could also have it error out with input not assigned or something...

donggyukim, 2016-01-11:
Let's return 0 in this case since nobody complains it.

ucbjrl, 2016-01-14:
I would argue for issuing a more explicit warning, and at least, randomizing the returned value (as we do for "uninitialized" values in general).

PR #635 (2016-01-22): Fix vcd dump
This will fix #623, #627
commit f7f13661 (2016-01-22): fix #623: print warning and return random
closed 2016-01-22 by ucbjrl