ucb-bar/chisel2-deprecated #633: More Mem questions

shunshou, 2016-01-19:
What is the difference between --backend fpga and --backend v?

I was expecting that --backend v would black box the Chisel mems (generated verilog would just have modules + port declarations that you'd fill in for mems) so that you could swap in mems from some ASIC mem compiler whereas --backend fpga would just inline the mem with appropriate Verilog constructs, since that was understandable by FPGA tools as BRAM.

Instead, I noticed that when using Mem, --backend fpga and --backend v both produce inline verilog mem logic... When using SeqMem, --backend fpga generates a *.conf file and mem blackboxes, whereas --backend v just produces inline logic again...

Am I just misinterpreting what the options are supposed to be for? 

I feel like --backend fpga should have isInlineMem = true instead of false?

Additionally, going back to the what is what mem debate... I was expecting that SeqRead for mem meant that the data output was registered, although the current implementation registers the read address... 

My motivation for registering the data out is that, at least for ASIC SRAM designs, the sense amp for the read out is clocked... so I've designed my read/right conflict resolution around that... (but since I haven't actually done any post synthesis verification, I don't really know if I'm doing the wrong thing...). Since my custom logic doesn't expect the way things are done w/ SeqMem (and gives me wrong outputs when i enable debugMem), should I be making my own MemOutReg that extends Mem, where seqRead = false and just register the data out when I do a mem.read operation? And hope that the blackbox assumes output registering in the compiled mem (for asic) while the fpga logic is completely inlined?

I'm wondering what others have done and whether they needed to change the way they did stuff after Mem was changed...

da-steve101, 2016-01-20:
Is [this](https://github.com/ucb-bar/chisel/blob/master/src/main/scala/FPGA.scala) what you are looking for? As you indicate above the only difference is the "isInlineMem" flag. As to if it should be true or false, I think that can be debated (I agree with you however). I personally just use the verilog backend and just pass flags I need. You can set it with "--inlineMem" or "--noInlineMem". You can see the defaults [here](https://github.com/ucb-bar/chisel/blob/master/src/main/scala/Driver.scala) at line 272. As for your last few qs, not sure, ill leave it to someone else to answer.

shunshou, 2016-01-20:
Yeah, I've already seen both the defaults and what exactly FPGA backend does. I'm just questioning the choice to have isLineMem be false for FPGA backend and true as default. Maybe it's just not worth having an FPGA backend option if it's questionable that isLineMem = false is actually suited for FPGA verilog. 

So I asked Stevo and Ben about ASIC mems, and while it depends on the specific technology, it's not a requirement that read addresses be registered. So then I wonder why only SeqMem and not Mem (with seqRead = false) supports black boxed memories? -- which would always be necessary for using ASIC SRAMs.

shunshou, 2016-01-21:
Also, I'd like to clarify something (that might be incorrectly implemented?) When you're debugging read/write conflicts, I think write should always have the same functionality, regardless of sequential or non-sequential read. 

Therefore seqRead = true should not have:

doWrite(Reg(next=addr), Reg(next=cond), Reg(next=data), None)
doWrite(addr, cond, random_data, None)

and instead only have:

doWrite(addr, cond, random_data, None)

to accurately model how the (supposed) memory functions. If anything, you should be returning a random number on read without actually affecting what is written to memory...

  def write(addr: UInt, dataIn: T): Unit = {
    val cond = Module.current.whenCond
    val data = dataIn.toBits
    if (seqRead && Driver.isDebugMem) {
      // generate bogus data when reading & writing same address on same cycle
      val random16 = LFSR16()
      val random_data = Cat(random16, Array.fill((needWidth()-1)/16){random16}:_*)
      doWrite(Reg(next=addr), Reg(next=cond), Reg(next=data), None)
      doWrite(addr, cond, random_data, None)
    } else {
      doWrite(addr, cond, data, None)
    }
  }