ucb-bar/chisel2-deprecated #634: Java Thread Deaths with Tester

cyarp, 2016-01-20:
I recall that this had been an issue with Java versions in the past.  This time, the JVM is Oracle 1.8 so I'm not sure if it is a version bug.  This is with Chisel 2.2.31 from maven.

```
RAN 2 CYCLES PASSED
[error] (Thread-11) java.lang.ThreadDeath
[error] (Thread-12) java.lang.ThreadDeath
java.lang.ThreadDeath
	at java.lang.Thread.stop(Thread.java:850)
	at scala.sys.process.ProcessImpl$SimpleProcess.destroy(ProcessImpl.scala:226)
	at Chisel.Tester.finish(Tester.scala:734)
	at Chisel.Driver$.apply(Driver.scala:53)
	at Chisel.chiselMain$.apply(hcl.scala:63)
	at Chisel.chiselMainTest$.apply(hcl.scala:76)
	at sha3.chiMain$.main(chi.scala:134)
	at sha3.chiMain.main(chi.scala)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
[trace] Stack trace suppressed: run last compile:run for the full output.
java.lang.ThreadDeath
	at java.lang.Thread.stop(Thread.java:850)
	at scala.sys.process.ProcessImpl$SimpleProcess.destroy(ProcessImpl.scala:226)
	at Chisel.Tester.finish(Tester.scala:734)
	at Chisel.Driver$.apply(Driver.scala:53)
	at Chisel.chiselMain$.apply(hcl.scala:63)
	at Chisel.chiselMainTest$.apply(hcl.scala:76)
	at sha3.chiMain$.main(chi.scala:134)
	at sha3.chiMain.main(chi.scala)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
[trace] Stack trace suppressed: run last compile:run for the full output.
[success] Total time: 14 s, completed Jan 19, 2016 4:46:33 PM
```

```
java -version
java version "1.8.0_66"
Java(TM) SE Runtime Environment (build 1.8.0_66-b17)
Java HotSpot(TM) 64-Bit Server VM (build 25.66-b17, mixed mode)
```

shunshou, 2016-01-20:
FYI John also saw it on a BWRC machine (which was updated to 1.7 to fix the same error a month ago...). I don't seem to have any problems running on my Mac though...

Original error I saw (with 1.6):

[error] (Thread-24) java.lang.ThreadDeath
[error] (Thread-23) java.lang.ThreadDeath
java.lang.ThreadDeath
        at java.lang.Thread.stop(Thread.java:836)
        at scala.sys.process.ProcessImpl$SimpleProcess.destroy(ProcessImpl.scala:226)
        at Chisel.Tester.finish(Tester.scala:636)
        at Chisel.Driver$.apply(Driver.scala:53)
        at Chisel.chiselMain$.apply(hcl.scala:63)
        at Chisel.chiselMainTest$.apply(hcl.scala:76)
        at controlUtils.PriorityArbiterMain$.main(controlUtils.scala:488)
        at controlUtils.PriorityArbiterMain.main(controlUtils.scala)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:57)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.lang.reflect.Method.invoke(Method.java:606)
[trace] Stack trace suppressed: run last compile:run for the full output.
java.lang.ThreadDeath
        at java.lang.Thread.stop(Thread.java:836)
        at scala.sys.process.ProcessImpl$SimpleProcess.destroy(ProcessImpl.scala:226)
        at Chisel.Tester.finish(Tester.scala:636)
        at Chisel.Driver$.apply(Driver.scala:53)
        at Chisel.chiselMain$.apply(hcl.scala:63)
        at Chisel.chiselMainTest$.apply(hcl.scala:76)
        at controlUtils.PriorityArbiterMain$.main(controlUtils.scala:488)
        at controlUtils.PriorityArbiterMain.main(controlUtils.scala)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:57)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.lang.reflect.Method.invoke(Method.java:606)
[trace] Stack trace suppressed: run last compile:run for the full output.
[success] Total time: 5 s, completed Dec 3, 2015 4:57:19 PM

cyarp, 2019-11-27:
Cleaning up old issues in my github.  I assume things have progressed a bit since early 2016 and this was either fixed or will never be fixed.

closed 2019-11-27 by cyarp