ucb-bar/chisel2-deprecated #650: Negative literal problem

ascenium, 2016-02-08:
The statement:

class foo extends Module
{
    val io = new Bundle
    {
        val out2 = UInt( OUTPUT, 32 )
    }
    out2 := UInt( 0xffffffff, 32 )
}

Yields the error message "UInt can't represent negative literal -1 in class foo"

Apparently, it can't represent the positive number 4,294,967,296 either.

aswaterman, 2016-02-09:
This is a Scala thing.  0xffffffff is of type Int, which is a 32-bit signed
type and so has value -1.

There are many workarounds including using BigInt, e.g. by writing
UInt((BigInt(1) << 32) - 1).

closed 2016-02-09 by aswaterman

ascenium, 2016-02-09:
Thanks for the workaround.  I guess this is why a language like scala (via Java) with no fundamental unsigned types might not be the ideal basis for a language that represents hardware.

ccelio, 2016-02-10:
> I guess this is why a language like scala (via Java) with no fundamental unsigned types might not be the ideal basis for a language that represents hardware.

You're just punting the problem a little further down the field - what are you suppose to do if you want to represent 33-bits? Or 65-bits? No matter what language is used as the base, you're going to hit edge cases when you rely on the native integer format types to specify literals. 

Chisel's UInt type accepts Scala Ints, BigInts, and Strings.  `UInt( "hffffffff", 32 )` should work just fine for your needs.  And if you find yourself needing to represent 33-bit literals, `UInt("h1ffffffff",33)` follows naturally.

ascenium, 2016-02-10:
Thanks for the much more readable and maintainable approach to specifying unsigned numbers in Chisel!  My hostility towards scala in my previous post was undeserved, and probably has more to do with past episodes of hair-ripping frustration at porting various bit manipulations from C++ to Java.  Like with learning any language, there is the process (that I'm in the middle of) where you're learning to be articulate enough in the new language to not paint yourself into ghastly corners.

ccelio, 2016-02-10:
No worries! =)