ucb-bar/chisel2-deprecated #658: tutorial hello asserts sim_api.h:37: channel_t::channel_t(int): Assertion  `channel != ((void *) -1)' failed.

timsifive, 2016-02-17:
I'm trying to get started with chisel, so I cloned chisel-tutorial. When I run 'make' in hello/, I get the following assertion and then sbt/chisel hangs:
```Hello: /media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello//sim_api.h:37: channel_t::channel_t(int): Assertion `channel != ((void *) -1)' failed.```

This is using Debian 8.3, and sbt 0.13.8.

Here's the full output of 'make':
```
set -e -o pipefail; "sbt" -Dsbt.log.noformat=true -DchiselVersion="latest.release" "run Hello --genHarness --compile --test --backend c --vcd --targetDir /media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello " | tee /media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello/Hello.out
[info] Loading project definition from /media/sf_tnewsome/Synced/SiFive/chisel-tutorial/hello/project
[info] Updating {file:/media/sf_tnewsome/Synced/SiFive/chisel-tutorial/hello/project/}hello-build...
[info] Resolving org.scala-lang#scala-library;2.10.4 ...
[info] Resolving org.scala-sbt#sbt;0.13.8 ...
[info] Resolving org.scala-sbt#main;0.13.8 ...
[info] Resolving org.scala-sbt#actions;0.13.8 ...
[info] Resolving org.scala-sbt#classpath;0.13.8 ...
[info] Resolving org.scala-lang#scala-compiler;2.10.4 ...
[info] Resolving org.scala-lang#scala-reflect;2.10.4 ...
[info] Resolving org.scala-sbt#launcher-interface;0.13.8 ...
[info] Resolving org.scala-sbt#interface;0.13.8 ...
[info] Resolving org.scala-sbt#io;0.13.8 ...
[info] Resolving org.scala-sbt#control;0.13.8 ...
[info] Resolving org.scala-sbt#completion;0.13.8 ...
[info] Resolving org.scala-sbt#collections;0.13.8 ...
[info] Resolving jline#jline;2.11 ...
[info] Resolving org.scala-sbt#api;0.13.8 ...
[info] Resolving org.scala-sbt#compiler-integration;0.13.8 ...
[info] Resolving org.scala-sbt#incremental-compiler;0.13.8 ...
[info] Resolving org.scala-sbt#logging;0.13.8 ...
[info] Resolving org.scala-sbt#process;0.13.8 ...
[info] Resolving org.scala-sbt#relation;0.13.8 ...
[info] Resolving org.scala-sbt#compile;0.13.8 ...
[info] Resolving org.scala-sbt#classfile;0.13.8 ...
[info] Resolving org.scala-sbt#persist;0.13.8 ...
[info] Resolving org.scala-tools.sbinary#sbinary_2.10;0.4.2 ...
[info] Resolving org.scala-sbt#compiler-ivy-integration;0.13.8 ...
[info] Resolving org.scala-sbt#ivy;0.13.8 ...
[info] Resolving org.scala-sbt#cross;0.13.8 ...
[info] Resolving org.scala-sbt.ivy#ivy;2.3.0-sbt-fccfbd44c9f64523b61398a0155784dcbaeae28f ...
[info] Resolving com.jcraft#jsch;0.1.46 ...
[info] Resolving org.scala-sbt#serialization_2.10;0.1.1 ...
[info] Resolving org.scala-lang.modules#scala-pickling_2.10;0.10.0 ...
[info] Resolving org.scalamacros#quasiquotes_2.10;2.0.1 ...
[info] Resolving org.json4s#json4s-core_2.10;3.2.10 ...
[info] Resolving org.json4s#json4s-ast_2.10;3.2.10 ...
[info] Resolving com.thoughtworks.paranamer#paranamer;2.6 ...
[info] Resolving org.spire-math#jawn-parser_2.10;0.6.0 ...
[info] Resolving org.spire-math#json4s-support_2.10;0.6.0 ...
[info] Resolving org.scala-sbt#run;0.13.8 ...
[info] Resolving org.scala-sbt#task-system;0.13.8 ...
[info] Resolving org.scala-sbt#tasks;0.13.8 ...
[info] Resolving org.scala-sbt#tracking;0.13.8 ...
[info] Resolving org.scala-sbt#cache;0.13.8 ...
[info] Resolving org.scala-sbt#testing;0.13.8 ...
[info] Resolving org.scala-sbt#test-agent;0.13.8 ...
[info] Resolving org.scala-sbt#test-interface;1.0 ...
[info] Resolving org.scala-sbt#main-settings;0.13.8 ...
[info] Resolving org.scala-sbt#apply-macro;0.13.8 ...
[info] Resolving org.scala-sbt#command;0.13.8 ...
[info] Resolving org.scala-sbt#logic;0.13.8 ...
[info] Resolving org.scala-sbt#compiler-interface;0.13.8 ...
[info] Resolving org.scala-sbt#precompiled-2_8_2;0.13.8 ...
[info] Resolving org.scala-sbt#precompiled-2_9_2;0.13.8 ...
[info] Resolving org.scala-sbt#precompiled-2_9_3;0.13.8 ...
[info] Resolving org.scala-lang#jline;2.10.4 ...
[info] Resolving org.fusesource.jansi#jansi;1.4 ...
[info] Done updating.
[info] Set current project to chisel-tutorial (in build file:/media/sf_tnewsome/Synced/SiFive/chisel-tutorial/hello/)
[info] Updating {file:/media/sf_tnewsome/Synced/SiFive/chisel-tutorial/hello/}hello...
[info] Resolving org.scala-lang#scala-library;2.11.7 ...
[info] Resolving edu.berkeley.cs#chisel_2.11;latest.release ...
[info] Resolving edu.berkeley.cs#chisel_2.11;latest.release ...
[info] Resolving org.scala-lang#scala-reflect;2.11.7 ...
[info] Resolving org.scala-lang#scala-compiler;2.11.7 ...
[info] Resolving org.scala-lang.modules#scala-xml_2.11;1.0.4 ...
[info] Resolving org.scala-lang.modules#scala-parser-combinators_2.11;1.0.4 ...
[info] Resolving jline#jline;2.12.1 ...
[info] Done updating.
[info] Compiling 1 Scala source to /media/sf_tnewsome/Synced/SiFive/chisel-tutorial/hello/target/scala-2.11/classes...
[info] Running Hello.Hello Hello --genHarness --compile --test --backend c --vcd --targetDir /media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello
CPP elaborate
[[35minfo[0m] [0.054] // COMPILING < (class Hello.Hello)>(0)
[[35minfo[0m] [0.087] giving names
[[35minfo[0m] [0.105] executing custom transforms
[[35minfo[0m] [0.106] convert masked writes of inline mems
[[35minfo[0m] [0.107] adding clocks and resets
[[35minfo[0m] [0.113] inferring widths
[[35minfo[0m] [0.125] checking widths
[[35minfo[0m] [0.126] lowering complex nodes to primitives
[[35minfo[0m] [0.126] removing type nodes
[[35minfo[0m] [0.129] compiling 2 nodes
[[35minfo[0m] [0.129] computing memory ports
[[35minfo[0m] [0.131] resolving nodes to the components
[[35minfo[0m] [0.170] creating clock domains
[[35minfo[0m] [0.173] pruning unconnected IOs
[[35minfo[0m] [0.178] checking for combinational loops
[[35minfo[0m] [0.187] NO COMBINATIONAL LOOP FOUND
[[35minfo[0m] [0.257] populating clock domains
CppBackend::elaborate: need 0, redundant 0 shadow registers
[[35minfo[0m] [0.282] generating cpp files
CppBackend: createCppFile Hello.cpp
[[35minfo[0m] [1.188] g++ -c -o /media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello//Hello-emulator.o  -I../ -I/csrc/  /media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello//Hello-emulator.cpp RET 0
[[35minfo[0m] [1.835] g++ -c -o /media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello//Hello.o  -I../ -I/csrc/  /media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello//Hello.cpp RET 0
[[35minfo[0m] [1.970] g++   -o /media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello//Hello /media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello//Hello.o /media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello//Hello-emulator.o RET 0
SEED 1455734905507
STARTING /media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello//Hello
Hello: /media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello//sim_api.h:37: channel_t::channel_t(int): Assertion `channel != ((void *) -1)' failed.
<no startup message>
make: *** Deleting file '/media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello/Hello.out'
../suffix.mk:83: recipe for target '/media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello/Hello.out' failed
make: *** [/media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello/Hello.out] Interrupt
```

Thanks for looking at this.

ucbjrl, 2016-02-17:
It looks like the mmap() call is returning an error. Debugging this is tricky. I'll generate a recipe.

ucbjrl, 2016-02-17:
cd to the generated directory (/media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello)
edit/patch the sim_api.h file as follows:

    % diff -u sim_api.h.old sim_api.h
    --- sim_api.h.old       2016-02-17 12:17:04.266769221 -0800
    +++ sim_api.h   2016-02-17 12:20:20.605665665 -0800
    @@ -13,6 +13,8 @@
     #include <unistd.h>
     #include <sys/mman.h>
     #include <time.h>
    +#include <error.h>
    +#include <errno.h>
     
     enum SIM_CMD { RESET, STEP, UPDATE, POKE, PEEK, FORCE, GETID, GETCHK, SETCLK, FIN };
     
    @@ -34,6 +36,9 @@
     assert(write(fd, "", 1) != -1);
     assert(fsync(fd) != -1);   // ensure the data is available
     channel = (char*)mmap(NULL, pgsize, PROT_READ | PROT_WRITE, MAP_SHARED, fd, 0); 
    +    if (channel == MAP_FAILED) {
    +        error(0, errno, "mmap failed for channel %d", fd);
    +    }
     assert(channel != MAP_FAILED);
       }
       ~channel_t() {

compile the simulator

    g++ -c -o /media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello//Hello-emulator.o  -I../ -I/csrc/  /media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello//Hello-emulator.cpp
    g++ -c -o /media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello//Hello.o  -I../ -I/csrc/  /media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello//Hello.cpp
    g++   -o /media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello//Hello /media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello//Hello.o /media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello//Hello-emulator.o

- launch the simulator

    /media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello/Hello

Although the simulator will hang, you should see the error message displayed.

timsifive, 2016-02-18:
Thanks for the quick reply! The error message is:
/media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello/Hello:
mmap failed for channel 3: Invalid argument

Some more info:

(gdb) bt
\#0  channel_t::channel_t (this=0x8056670, _fd=3)
    at /media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello//sim_api.h:39
\#1  0x0804e5a1 in sim_api_t<dat_api_base*>::sim_api_t (this=0xbffff6c4)
    at /media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello//sim_api.h:82
\#2  0x0804ddde in emul_api_t::emul_api_t (this=0xbffff6c4, m=0xbffff688)
    at /media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello//emul_api.h:87
\#3  0x0804e0d9 in Hello_api_t::Hello_api_t (this=0xbffff6c4, m=0xbffff688)
    at /media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello//Hello.h:31
\#4  0x0804d980 in main (argc=1, argv=0xbffff7f4)
    at /media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello//Hello-emulator.cpp:5
(gdb) p pgsize
$1 = 4096
(gdb) p fd
$2 = 3

So I assume it’s just unhappy about the fd.

tnewsome@toy-debian:/media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello$
sudo lsof -p 5989
COMMAND  PID     USER   FD   TYPE DEVICE SIZE/OFF   NODE NAME
Hello   5989 tnewsome  cwd    DIR   0,36     4096  34375
/media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello
Hello   5989 tnewsome  rtd    DIR    8,1     4096      2 /
Hello   5989 tnewsome  txt    REG   0,36   296468  41271
/media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello/Hello
Hello   5989 tnewsome  mem    REG    8,1  1750780 786922
/lib/i386-linux-gnu/i686/cmov/libc-2.19.so
Hello   5989 tnewsome  mem    REG    8,1   114964 789775
/lib/i386-linux-gnu/libgcc_s.so.1
Hello   5989 tnewsome  mem    REG    8,1   280108 786926
/lib/i386-linux-gnu/i686/cmov/libm-2.19.so
Hello   5989 tnewsome  mem    REG    8,1   963316 654692
/usr/lib/i386-linux-gnu/libstdc++.so.6.0.20
Hello   5989 tnewsome  mem    REG    8,1   134380 785073
/lib/i386-linux-gnu/ld-2.19.so
Hello   5989 tnewsome    0u   CHR  136,3      0t0      6 /dev/pts/3
Hello   5989 tnewsome    1u   CHR  136,3      0t0      6 /dev/pts/3
Hello   5989 tnewsome    2u   CHR  136,3      0t0      6 /dev/pts/3
Hello   5989 tnewsome    3u   REG   0,36     4096  41272
/media/sf_tnewsome/Synced/SiFive/chisel-tutorial/generated/hello/00005989.in

Which all looks kosher, but I think I know what the problem is…
*quick test*
Yeah, the issue is that /media/sf_tnewsome is mounted on vboxsf (which in
turn is sharing an ntfs). If I move the whole thing to an ext4 file system
then it works fine.
Thanks for your help! In the future it’d be cool if the error message could
include the fd as well as the path of the file being mmapped, which is what

Tim
​

ucbjrl, 2016-02-19:
Closed by #660

closed 2016-02-19 by ucbjrl