ucb-bar/chisel2-deprecated #665: Segmentation fault crashing tester

da-steve101, 2016-02-26:
The tester crashed with a segmentation fault

Chisel trace is
```bash
[info]   Chisel.TestApplicationException: test application exit - exit code 139
[info]   at Chisel.Tester.throwExceptionIfDead(Tester.scala:128)
[info]   at Chisel.Tester.Chisel$Tester$$mwhile(Tester.scala:138)
[info]   at Chisel.Tester.start(Tester.scala:763)
[info]   at Chisel.Tester.<init>(Tester.scala:795)
```

Valgrind output running the simulator is
```bash
==22869== Invalid write of size 8
==22869==    at 0x7A8482: dat_api<17>::get_value(unsigned long*) (emul_api.h:38)
==22869==    by 0x7AA7CB: emul_api_t::get_value(dat_api_base*&, unsigned long*) (emul_api.h:110)
==22869==    by 0x7F25AF: sim_api_t<dat_api_base*>::send_tokens() (sim_api.h:333)
==22869==    by 0x7F21B0: sim_api_t<dat_api_base*>::tick() (sim_api.h:142)
```

From gdb looking at sim_api.h (send_tokens, 325) at the iteration of loop at 331 before the crash
```bash
sim_data.outputs.size() = 5367
off = 2047
```

It appears that the out channel is not big enough for all the values. The channel appears to allocate one page (pgsize = 4096) to store values (sim_api.h, 55).
```c
    channel = (char*)mmap(NULL, pgsize, PROT_READ | PROT_WRITE, MAP_SHARED, fd, 0); 
```
Perhaps the channel size should be determined using the number of inputs and outputs the design has?

ucbjrl, 2016-02-26:
On Feb 25, 2016, at 5:48 PM, da-steve101 <notifications@github.com> wrote:

> The tester crashed with a segmentation fault
> 
> Chisel trace is
> 
> [info]   Chisel.TestApplicationException: test application exit - exit
>  code 139
> [info]   at Chisel.Tester.throwExceptionIfDead(Tester.scala:128)
> [info]   at Chisel.Tester.Chisel
> $Tester$$
> mwhile(Tester.scala:138)
> [info]   at Chisel.Tester.start(Tester.scala:763)
> [info]   at Chisel.Tester.
> <init>(Tester.scala:795)
> Valgrind output running the simulator is
> 
> ==22869== Invalid write of size 8
> ==22869==    at 0x7A8482: dat_api
> <17>::get_value(unsigned long*
> ) (emul_api.h:38)
> ==22869==    by 0x7AA7CB: emul_api_t::get_value(dat_api_base
> *&, unsigned long*
> ) (emul_api.h:110)
> ==22869==    by 0x7F25AF: 
> sim_api_t<dat_api_base*>::send_tokens
> () (sim_api.h:333)
> ==22869==    by 0x7F21B0: 
> sim_api_t<dat_api_base*>::tick() (sim_api.h:142)
> From gdb looking at sim_api.h (send_tokens, 325) at the iteration of loop at 331 before the crash
> 
> sim_data.outputs.size
> () = 5367
> off = 2047
> 
> It appears that the out channel is not big enough for all the values. The channel appears to allocate one page (pgsize = 4096) to store values (sim_api.h, 55).
> 
>     channel = (char*)mmap(NULL, pgsize, PROT_READ | PROT_WRITE, MAP_SHARED, fd, 0); 
> Perhaps the channel size should be determined using the number of inputs and outputs the design has?
> 

Excellent analysis. We'll submit a fix.

swamy-t, 2016-11-25:
Can someone comment on how to run valgrind or gdb with the C++ generated code? Normally I use sbt to run the binary file under the Chisel tester so I am not how to run both under valgrind/gdb in order to debug a segmentation fault. Thanks!

da-steve101, 2016-11-30:
When you run the chisel, it prints out the gcc commands it is running. All you have to do is run those commands with '-g' to put in debugging symbols. Can then run the compiled binary with gdb/valgrind