ucb-bar/chisel2-deprecated #675: Make Printf Cycle Accurate

da-steve101, 2016-03-12:
As discussed in #671, I think printf should be cycle accurate.
Consider the following two examples of printf usage:
```scala
class MyMod extends Module {
  val io = new Bundle {
    val enable = Bool(INPUT)
    val a = UInt(INPUT, 4)
    val b = UInt(INPUT, 4)
    val c = UInt(OUTPUT, 4)
  }
  io.c := UInt(0, 4)
  when( io.enable ) {
    printf("MyMod add enabled\n")
    io.c := io.a + io.b
  }
}
```
```scala
class MyMod( flag : Boolean ) extends Module {
  val io = new Bundle {
    val a = UInt(INPUT, 4)
    val b = UInt(INPUT, 4)
    val c = UInt(OUTPUT, 4)
  }
  if ( flag ) {
    val myReg = RegNext( io.a + io.b )
    io.c := myReg
    printf("myReg = %d\n", myReg)
  } else {
    io.c := io.a - io.b
  }
}
```
In the first case, if trying to implement a conditional printf, you would have to duplicate the logic in the tester.
In the second case myReg isn't visible outside the if statement and hence cannot be peeked.

donggyukim, 2016-03-16:
Try it with #682. Let me know it's still broken. We should make tester sleep if prints do not pop up immediately, which is not desirable at all.

da-steve101, 2016-03-17:
Seems to be working :) thanks

closed 2016-03-17 by da-steve101