ucb-bar/chisel2-deprecated #677: Verilog Address Widths Incorrect

JackDavidson, 2016-03-12:
The Verilog exported from CHISEL has explicitly set widths for addresses into inputs, outputs and wires. The addresses seem to be coming from the wrong place, though. Rather than being determined by the width of the wire which is being extracted from, they seem to be set to whatever the minimum with of  the number is. Below is an example:

module LUT(
    input [3:0] io_lut,
    input [1:0] io_sel,
    output io_res
);
  wire muxb;
  wire T0;
  wire[1:0] muxa;
  wire[1:0] T1;
  wire[1:0] T2;
  wire T3;
  wire T4;
  wire T5;

  assign io_res = muxb;
  assign muxb = T5 ? T4 : T0;
  assign T0 = muxa[1'h0]; 
  assign muxa = T3 ? T2 : T1;
  assign T1 = io_lut[1'h1:1'h0];
  assign T2 = io_lut[2'h3:2'h2];
  assign T3 = io_sel[1'h1];
  assign T4 = muxa[1'h1];
  assign T5 = io_sel[1'h0];
endmodule

The result is only generating a warning, but its still something that should be fixed. If someone can point me in the general

da-steve101, 2016-03-13:
This verilog is still functionally correct though right?
As for where it is, the method to extract bits "io.lut(3, 2)" is defined in Bits.scala line 349
This uses extract which is in "Extract.scala"
As for the part you are interested in, "Verilog.scala" is where verilog is generated.
Lines 250 - 276 is where extract is dealt with

JackDavidson, 2016-03-14:
Yes, the verilog still functions fine :)

The reason I want to deal with this is because verilator does not like the explicitly set widths that are actually smaller than they should be. You can disable the warning in verilator, but it would be best to fix it entirely if the solution is simple.

I'll go take a look at the code. Thanks for your help!

JackDavidson, 2016-03-14:
So I looked into it and added a couple new functions. I have everything working on my machine, and I'm just waiting for the CHISEL tests to complete before I make a branch on git hub.

It looks like there is a value, "extractWidth" in the emitDef function that is never used. I'm fairly certain that it was originally intended to do exactly what I'm doing. Since it doesn't seem to be used for anything else, I used it to do this.

A quick question though: is there any reason in particular that we use hex numbers with specified widths rather than simply using base 10 numbers?

da-steve101, 2016-03-14:
Not sure on the hex vs decimal.
I would be surprised if the tests pass as the verilog comparator does a direct comparison.
The verilog compared against is in src/test/resources and is compared against the verilog generated in a folder at the top level called test-outputs/

JackDavidson, 2016-03-14:
Right after I posted that, I found that you're right. Some tests are failing because of comparisons with pre-generated verilog or C++ code

It looks like 100% of the test failures are happening in "assertFile( filename: String )" in the file "TestHelper.scala". When I change the first line in the function to just return immediately, I get all passes.

so it doesn't look like the change is interfering with anything else. I'll manually check the files that are failing before I modify the tests, hopefully tomorrow

JackDavidson, 2016-03-14:
O.K. I'm going to do a little extra tests on my own, but I have the changes up at: 

https://github.com/JackDavidson/chisel/tree/fixBitWithOfVerilogIndices

Would someone mind looking it over to see if I'm doing things the preferred way?
Its only a few lines of code.

da-steve101, 2016-03-15:
You can just use log2Up() instead but other than that looks ok to me

JackDavidson, 2016-03-15:
My commit that changes the index constants to base 10 was merged, so I'm closing this :)

closed 2016-03-15 by JackDavidson