ucb-bar/chisel2-deprecated #689: Chisel.TestApplicationException: test application exit - exit code 139

ascenium, 2016-04-11:
I'm also being stopped by exit code 139 problems -- translates to an illegal memory access by the C++ simulation.

My Chisel trace is as follows:

Chisel.TestApplicationException: test application exit - exit code 139
	at Chisel.Tester.Chisel$Tester$$mwhile(Tester.scala:131)
	at Chisel.Tester$$anonfun$start$3.apply$mcVI$sp(Tester.scala:747)
	at scala.collection.immutable.Range.foreach$mVc$sp(Range.scala:166)
	at Chisel.Tester.start(Tester.scala:745)
	at Chisel.Tester.<init>(Tester.scala:775)
	at ascenium.ncprocessor_Tests.<init>(ncprocessor.scala:369)
	at ascenium.main$$anonfun$main$2.apply(main.scala:106)
	at ascenium.main$$anonfun$main$2.apply(main.scala:106)
	at Chisel.Driver$.apply(Driver.scala:65)
	at Chisel.chiselMain$.apply(hcl.scala:63)
	at Chisel.chiselMainTest$.apply(hcl.scala:76)
	at ascenium.main$.main(main.scala:105)
	at ascenium.main.main(main.scala)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:57)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:606)

Looks like the tester is getting a lot farther than in the other posted exit 139 crash, so probably unrelated to it.

Please let me know what else you need from me to track down the problem.

da-steve101, 2016-04-11:
Can you compile the cpp binary with "-g" and run in gdb to find the section of code?
Please post that section of code (and maybe attach the cpp files too)

ascenium, 2016-04-12:
Will do.

ascenium, 2016-05-06:
When I do this, the compiled code runs forever and apparently does nothing.  It doesn't crash, so there's no "location" of bad code I can send you.

I expect that this problem is a timeout problem where the gcc compiler is somehow in a race condition with the Chisel debugger.  In instances where the generated model is too large to compile in time, you get this error.

da-steve101, 2016-05-07:
I have had a similar issue in the past, the c is waiting for input from the debugger (ie reset/step etc). just sending any string is usually enough to trigger the crash. Could you attach the cpp (the emulator one too) and ill see if I can replicate it?