ucb-bar/chisel2-deprecated #692: Testers Vec size issue: java.lang.IndexOutOfBoundsException

hqjenny, 2016-04-13:
We are working on our CS250 project with many wires connections and a parameter to control the number of row modules. When we increase number of rows to a specific number(e.g. 8), java.lang.IndexOutOfBoundsException will show up. The number of I/Os is a multiple of number of rows. 

We tried to find the issue using a dummy modules shown as below. Once the parameter S is increased to 512, Chisel will throw IndexOutOfBoundsException at runtime when we run make-run-unit. There seems to be an upper bound for parameter W too. When S is set to 511, W can only be set up to 65, otherwise the same exception will show up. 

We are using  scalaVersion := "2.11.6", sbt.version="0.13.8". 

Source Code:
```
package sha3

import Chisel._

class SizeModule(val S: Int = 511, val W: Int = 1) extends Module {
  val io = new Bundle { 
    val inbits = Vec.fill(S){Bits(INPUT,W)}
    val outbits = Vec.fill(S){Bits(OUTPUT,W)}
  }
  for(i <- 0 until S) {
  	io.outbits(i) := io.inbits(i)
  }
}

class SizeModuleTests(c: SizeModule) extends Tester(c) {
      step(1)
}

object sizeMain { 
  def main(args: Array[String]): Unit = {
    //chiselMainTest(Array[String]("--backend", "c", "--genHarness", "--compile", "--test"),
    chiselMainTest(args,
    () => Module(new SizeModule())){c => new SizeModuleTests(c)
    }
  }
}

```
Error Trace:

ucbjrl, 2016-04-14:
This is due to a bug in the way Tester <-> Simulation communications memory is allocated. A fix for this has been incorporated into the master GitHub repo (pr #666), but hasn't made it into a release.

We should generate a new release within the next few days.

ucbjrl, 2016-04-15:
We've released Chisel v2.2.33 which should resolve this issue. Please let me know if it doesn't.