ucb-bar/chisel2-deprecated #707: Cautionary Tale -- Recommendations for checking tool compatibility (major Chisel/ISE issue)

shunshou, 2016-05-18:
So I've spent the last n days debugging someone else's Chisel/Verilog, which generally exhibited this symptom:

Verilog testbench passes in Vivado (behavioral + post synthesis)
Behavioral Verilog testbench passes in ISE 12.x-14.2
Post synthesis, Verilog testbench fails -- whole modules are optimized out during synthesis with lots of "TXXX<y> is removed b/c ____" warning messages (painful to debug; ISE is required for Virtex 5 support, which is what the NI SDR platform uses)

I eventually found the cause. See the following (valid) Verilog (simplified version of what Chisel emits) that fails post synthesis:

IO: input [4:0] io_simplecomparein,
wire [4:0] T722;
assign io_simplecompare = $signed(io_simplecomparein) < $signed(T722);
assign T722 = {2'h3, 3'h5};

----

Something that should be functionally equivalent (and actually passes post-synthesis simulation with ISE):

IO: input [4:0] io_simplecomparein
wire [4:0] T722;
assign T722 = {2'h3, 3'h5};
assign io_simplecompare = $signed(io_simplecomparein) < $signed(T722);

-----

Basically, the *signed* comparison was wrong when T722 was assigned *below* the < operation, even though it should be perfectly valid to assign T722 below the < operation. 

-----

This is really just an issue of how the emitted Verilog was misinterpreted by ISE. As a designer, it's really hard to catch such a problem, especially in someone else's code. I think it would make a lot of sense to have a dummy test suite module of common operations (that people could add to; + corresponding TB) that could be easily run to verify tool compatibility, lest you want to spend boatloads of time trying to figure out if it was your code that was incorrect or some silly tool bug...

Alternatively, how hard would it be to make sure that a generated Verilog signal is never used before it is actually assigned to? I assume most people (learn to) code from top to bottom... which might explain why this *bug* was only fixed between the jump from ISE to Vivado (rather than in the many ISE releases...)...

albert-magyar, 2016-05-18:
If it's that simple that's a really embarrassing bug in ISE.

shunshou, 2016-05-18:
I think it was probably something that wasn't caught until the Vivado upgrade b/c no one actually writes code like that... despite it being perfectly valid... >_> <_< 

1) I guess people who know what's going on with $signed wouldn't try to play it safe and manually sign extend (my bad; here's when being conservative about tool capabilities backfired...) -- this is still somewhat questionably problematic if I had two Fixed with mismatched fractional width (where one was also a lit)
2) If they did want to sign extend a constant, they'd just represent it with the right # of bits to begin with, rather than doing something like SInt(-3), which would give the representation with the least # of bits. 
3) If they started with a constant with minimal # of bits and then chose to sign extend, they would probably do it in a more logical way i.e. assign T722 before the comparison...

aswaterman, 2016-05-18:
Sounds like it would benefit everyone if you reported this to Xilinx.