ucb-bar/chisel2-deprecated #715: Will OrderedDecoupledHWIOTester insert bubble for the output.ready signal?

songmaotian, 2016-07-02:
Hello, 
  I am using the OrderedDecoupledHWIOTester for testing, I check the wavform, it seems the output.ready signal maintain high, Is it possible to insert random bubble for the signal, so that the test will be much better

chick, 2016-07-05:
Is it possible to provide some example code here?

songmaotian, 2016-07-06:
There's a example in src/test/scala/DecoupledGCD.scala

and I have found a bug, and make a quick fix as following

`diff --git a/src/main/scala/iotesters/OrderedDecoupledHWIOTester.scala b/src/main/scala/iotesters/OrderedDecoupledHWIOTester.scala
index 8673e88..6d1f054 100644
--- a/src/main/scala/iotesters/OrderedDecoupledHWIOTester.scala
+++ b/src/main/scala/iotesters/OrderedDecoupledHWIOTester.scala
@@ -262,6 +262,16 @@ abstract class OrderedDecoupledHWIOTester extends HWIOTester {
     )
   }
 
+  private def createOutputIsMyTurnTable(events: ArrayBuffer[TestingEvent]): Vec[Bool] = {
+    val associated_event_numbers = events.map { event => event.event_number }.toSet
+    logScalaDebug(s"  associated event numbers ${associated_event_numbers.toArray.sorted.mkString(",")}")
+
+    Vec(
+      output_event_list.indices.map { event_number => Bool(associated_event_numbers.contains(event_number)) } ++
+        List(Bool(false)) // We append a false at the end so no-one tries to go when counter done
+    )
+  }
+
   /**
     * build a set of all ports referenced by all events associated with a particular
     * io interface
@@ -333,7 +343,7 @@ abstract class OrderedDecoupledHWIOTester extends HWIOTester {
   private def buildDecoupledOutputEventHandlers(event_counter: GlobalEventCounter) {
     decoupled_control_port_to_output_values.foreach { case (controlling_port, events) =>
       val ports_referenced_for_this_controlling_port = portsReferencedByEvents(events)
-      val is_this_my_turn = createIsMyTurnTable(events)
+      val is_this_my_turn = createOutputIsMyTurnTable(events)
 
       val counter_for_this_decoupled = Counter(output_event_list.length)
       logScalaDebug(s"Output decoupled controller ${name(controlling_port)} : ports " +
-- 
1.9.1
`