ucb-bar/chisel2-deprecated #718: Vec.fill appears to be badly broken

ascenium, 2016-07-18:
Imagine our horror this weekend when we built our code with the latest version of Chisel and received a java.lang.IndexOutOfBoundsException on every one of our many dozens of Vec.fill statements.  This command worked as recently as June 30.  My Vec.fill commands are generally of a form similar to this example:

var i = 0
val x = Vec.fill( 32 )
{
    val y = z[i](0)
    i = i + 1
    y
}

...where z is a Vec 32 UInt INPUT of some kind.  It appears that in this case i is getting up to 32 or perhaps higher (though 32 is where the failure happens).  If you do this it compiles:

//var i = 0
val x = Vec.fill( 32 )
{
//    val y = z[i](0)
//    i = i + 1
//    y
Bool( false )
}

ucbjrl, 2016-07-18:
@ascenium could you confirm that you're using the current master 52f7e78, or release  4b40209 ?
In the meantime, we'll try to duplicate the issue.

ucbjrl, 2016-07-18:
We've verified the issue and we're testing a fix.

aswaterman, 2016-07-18:
Sounds like Vec.tabulate is more appropriate for your use case (not that this isn't a legitimate issue).

ascenium, 2016-07-18:
Per the earlier question, we found this on the version of Chisel we downloaded fresh this morning.  With regard to Vec.tabulate, I haven't seen it mentioned in the docs.  I've been using Chisel for about 9 months now, so I tend to re-use what I have determined is workable syntax unless I absolutely require something new.  If there is an email alert list for doc updates, I will gladly join and review.

ucbjrl, 2016-07-19:
Unfortunately, in this case, Vec.tabulate would exhibit the same issue.

ucbjrl, 2016-07-19:
Fixed by #721 (dedfa31)

closed 2016-07-19 by ucbjrl

ascenium, 2016-07-22:
How do I pull the release that fixes this bug?  "latest.release" doesn't cut it.

ascenium, 2016-07-22:
I tried "latest.integration" as well, no joy.

ucbjrl, 2016-07-22:
On Jul 21, 2016, at 5:45 PM, Robert Mykland <notifications@github.com> wrote:
> 
> How do I pull the release that fixes this bug? "latest.release" doesn't cut it.
> 

We haven't published a new release with the fix. We will do so over the weekend.

In the meantime, you have two options:
 - use version 2.2.33 (4/14/16)
 - clone https://github.com/ucb-bar/chisel.git, sbt publish-local, use version 2.3-SNAPSHOT

ucbjrl, 2016-07-23:
We've published 2.2.35. "latest.release" should now contain the fix.