ucb-bar/chisel2-deprecated #723: firrtl_interpreter/Concrete.scala: randomSInt produces out of bounds value.

stevenmburns, 2016-08-04:
I get this stack trace when running a medium sized design.

...
Exception during evaluation: error: ConcreteSInt(4140379555, 32) bad width 32 needs 33  @[Decoupled.scala 127:17]
Expression Evaluation stack
   0 fifo_diag_1_1.Queue_1_1.ram_sim_value.T_71.data:MemIO -> fifo_diag_1_1.Queue_1_1.GEN_3
   1 fifo_diag_1_1.Queue_1_1.GEN_3:Node -> validif(fifo_diag_1_1.Queue_1_1.do_enq, fifo_diag_1_1.Queue_1_1.io_enq_bits_sim_value)
[error] (run-main-0) firrtl_interpreter.InterpreterException: error: ConcreteSInt(4140379555, 32) bad width 32 needs 33
firrtl_interpreter.InterpreterException: error: ConcreteSInt(4140379555, 32) bad width 32 needs 33
	at firrtl_interpreter.ConcreteSInt.<init>(Concrete.scala:329)
	at firrtl_interpreter.Concrete$.randomSInt(Concrete.scala:285)
	at firrtl_interpreter.LoFirrtlExpressionEvaluator.evaluate(LoFirrtlExpressionEvaluator.scala:308)
	at firrtl_interpreter.LoFirrtlExpressionEvaluator$$anonfun$3.apply(LoFirrtlExpressionEvaluator.scala:413)
	at firrtl_interpreter.LoFirrtlExpressionEvaluator$$anonfun$3.apply(LoFirrtlExpressionEvaluator.scala:408)
	at firrtl_interpreter.Timer.apply(Timer.scala:40)
...

This should fix the issue, but I wasn't able to run a local build to check.

smburns@smburns-VirtualBox:~/SW/firrtl-interpreter$ git diff src
diff --git a/src/main/scala/firrtl_interpreter/Concrete.scala b/src/main/scala/firrtl_interpreter/Concrete
index 737dadc..5bab6eb 100644
--- a/src/main/scala/firrtl_interpreter/Concrete.scala
+++ b/src/main/scala/firrtl_interpreter/Concrete.scala
@@ -282,7 +282,7 @@ object Concrete {
     }
   }
   def randomUInt(width: Int): ConcreteUInt  = ConcreteUInt(randomBigInt(width), width)
-  def randomSInt(width: Int): ConcreteSInt  = ConcreteSInt(randomBigInt(width), width)
+  def randomSInt(width: Int): ConcreteSInt  = ConcreteSInt(randomBigInt(width)-(BigInt(1)<<(width-1)), wi
   def randomClock():          ConcreteClock = ConcreteClock(randomBigInt(1))
 }

stevenmburns, 2016-08-04:
This is the correct diff:

diff --git a/src/main/scala/firrtl_interpreter/Concrete.scala b/src/main/scala/firrtl_interpreter/Concrete.scala
index 737dadc..5bab6eb 100644
--- a/src/main/scala/firrtl_interpreter/Concrete.scala
+++ b/src/main/scala/firrtl_interpreter/Concrete.scala
@@ -282,7 +282,7 @@ object Concrete {
     }
   }
   def randomUInt(width: Int): ConcreteUInt  = ConcreteUInt(randomBigInt(width), width)
-  def randomSInt(width: Int): ConcreteSInt  = ConcreteSInt(randomBigInt(width), width)
+  def randomSInt(width: Int): ConcreteSInt  = ConcreteSInt(randomBigInt(width)-(BigInt(1)<<(width-1)), width)
   def randomClock():          ConcreteClock = ConcreteClock(randomBigInt(1))
 }

stevenmburns, 2016-08-04:
`diff --git a/src/main/scala/firrtl_interpreter/Concrete.scala b/src/main/scala/firrtl_interpreter/Concrete.scala
index 737dadc..5bab6eb 100644
--- a/src/main/scala/firrtl_interpreter/Concrete.scala
+++ b/src/main/scala/firrtl_interpreter/Concrete.scala
@@ -282,7 +282,7 @@ object Concrete {
     }
   }
   def randomUInt(width: Int): ConcreteUInt  = ConcreteUInt(randomBigInt(width), width)
-  def randomSInt(width: Int): ConcreteSInt  = ConcreteSInt(randomBigInt(width), width)
+  def randomSInt(width: Int): ConcreteSInt  = ConcreteSInt(randomBigInt(width)-(BigInt(1)<<(width-1)), width)
   def randomClock():          ConcreteClock = ConcreteClock(randomBigInt(1))
 }
 
`

stevenmburns, 2016-08-04:
Added the correct code fences:

````
diff --git a/src/main/scala/firrtl_interpreter/Concrete.scala b/src/main/scala/firrtl_interpreter/Concrete.scala
index 737dadc..5bab6eb 100644
--- a/src/main/scala/firrtl_interpreter/Concrete.scala
+++ b/src/main/scala/firrtl_interpreter/Concrete.scala
@@ -282,7 +282,7 @@ object Concrete {
     }
   }
   def randomUInt(width: Int): ConcreteUInt  = ConcreteUInt(randomBigInt(width), width)
-  def randomSInt(width: Int): ConcreteSInt  = ConcreteSInt(randomBigInt(width), width)
+  def randomSInt(width: Int): ConcreteSInt  = ConcreteSInt(randomBigInt(width)-(BigInt(1)<<(width-1)), width)
   def randomClock():          ConcreteClock = ConcreteClock(randomBigInt(1))
 }
````

stevenmburns, 2016-08-04:
Added the correct code fence:
````
diff --git a/src/main/scala/firrtl_interpreter/Concrete.scala b/src/main/scala/firrtl_interpreter/Concrete.scala
index 737dadc..5bab6eb 100644
--- a/src/main/scala/firrtl_interpreter/Concrete.scala
+++ b/src/main/scala/firrtl_interpreter/Concrete.scala
@@ -282,7 +282,7 @@ object Concrete {
     }
   }
   def randomUInt(width: Int): ConcreteUInt  = ConcreteUInt(randomBigInt(width), width)
-  def randomSInt(width: Int): ConcreteSInt  = ConcreteSInt(randomBigInt(width), width)
+  def randomSInt(width: Int): ConcreteSInt  = ConcreteSInt(randomBigInt(width)-(BigInt(1)<<(width-1)), width)
   def randomClock():          ConcreteClock = ConcreteClock(randomBigInt(1))
 }
 ````

chick, 2016-08-04:
I have opened an issue [firrtl-interpreter #18](https://github.com/ucb-bar/firrtl-interpreter/issues/18).  A fix will be coming shortly