ucb-bar/chisel2-deprecated #724: RegUpdate and RegReset?

cornytrace, 2016-08-15:
These functions are used in the tutorial and manual, but are not explained. A google search has also revealed nothing. Can someone explain what they do, and perhaps add this information to the manual?

da-steve101, 2016-08-31:
I can't find them in the chisel source or tutorial source. You sure you don't mean RegNext and RegInit?
RegNext( x ) creates a register directly after x. RegInit( x ) will reset to the value of x and other behaviour must be further defined.

cornytrace, 2016-09-07:
Look here: https://chisel.eecs.berkeley.edu/2.2.0/manual.html
Sections 7 and 8 use RegUpdate and RegReset

da-steve101, 2016-09-12:
Ok sure. That is really out of date. Changed around 3 yrs ago see commit https://github.com/ucb-bar/chisel/commit/79134837e243f071664048f634c9e17352b0c6de
Essentially RegReset has been renamed to RegInit and RegUpdate has been renamed RegNext