ucb-bar/chisel2-deprecated #725: Q: "make getting-started" fails

omerfirmak, 2016-08-18:
I am following the instructions in the following link.

https://chisel.eecs.berkeley.edu/2.2.0/getting-started.html

I run the "make Parity.out" command with success. But "make getting-started" fails no matter how many times i try it. I have seen at most 4 of the 16 tests pass.

I have gcc 6.1.1 , openjdk 8 on a Arch Linux system.

ucbjrl, 2016-08-18:
> On Aug 18, 2016, at 1:16 AM, Omer Faruk IRMAK <notifications@github.com> wrote:
> 
> I am following the instructions in the following link.
> 
> https://chisel.eecs.berkeley.edu/2.2.0/getting-started.html
> 
> I run the "make Parity.out" command with success. But "make getting-started" fails no matter how many times i try it. I have seen at most 4 of the 16 tests pass.
> 
> I have gcc 6.1.1 , openjdk 8 on a Arch Linux system.
> 

Are you editing the problem files to add the missing code elements?

They will fail until you do so.

omerfirmak, 2016-08-18:
@ucbjrl 
No, I haven't edited anything so far.
Can you point me to a link to help me with those?

ucbjrl, 2016-08-18:
> On Aug 18, 2016, at 8:38 AM, Omer Faruk IRMAK <notifications@github.com> wrote:
> 
> @ucbjrl 
> No, I haven't edited anything so far.
> Can you point me to a link to help me with those?
> 

Download and read the Chisel 2.2 Tutorial - https://chisel.eecs.berkeley.edu/2.2.0/chisel-tutorial.pdf.
Read the individual chisel-tutorial/problems Scala files and follow the instructions in the comments.

omerfirmak, 2016-08-21:
Makes sense now :)
Thanks.

closed 2016-08-21 by omerfirmak