ucb-bar/chisel2-deprecated #729: Exception: sbt.TrapExitSecurityException

drom, 2016-09-12:
Fresh git clone. Running `sbt run` getting this error:

```
drom@drom:~/work/github/usb-bar/chisel> ~/programs/sbt/bin/sbt run
[info] Loading project definition from /home/drom/work/github/usb-bar/chisel/project
[info] Set current project to chisel (in build file:/home/drom/work/github/usb-bar/chisel/)
[info] Running Chisel.SCWrapper 
Error writing file generated/generated.cpp

Exception: sbt.TrapExitSecurityException thrown from the UncaughtExceptionHandler in thread "run-main-0"
java.lang.RuntimeException: Nonzero exit code: -1
        at scala.sys.package$.error(package.scala:27)
[trace] Stack trace suppressed: run last compile:run for the full output.
[error] (compile:run) Nonzero exit code: -1
[error] Total time: 1 s, completed Sep 12, 2016 12:05:25 PM
```
Any guidance how to fix?

timothychong, 2016-11-24:
Can someone help with this?

ucbjrl, 2016-11-24:
We'll need a little more context to help here:

 - what is the host system

 - which version of sbt is installed

 - have you created a design in scala with a main method

I assume this is "chisel" (chisel2 as opposed to chisel3). In either case

    % sbt run

without a user created design will fail and is not the correct way to use
chisel.

If you are just starting out, we strongly suggest you start with chisel3
<https://github.com/ucb-bar/chisel3>. Start with the tutorial
<https://github.com/ucb-bar/chisel-tutorial>, then use template
<https://github.com/ucb-bar/chisel-template> to create a basic chisel
project.

There is documentation in the associated GitHub wikis to help you get
started.

If you really do want to use chisel2, we suggest you check out the chisel2
branch of the tutorial
<https://github.com/ucb-bar/chisel-tutorial/tree/chisel2> and verify that
it executes in your environment.