ucb-bar/chisel2-deprecated #739: libraryDependencies latest.release not found

colin4124, 2016-11-17:
I follow README.md:

`libraryDependencies += "edu.berkeley.cs" %% "chisel" % "latest.release"`

but I got this error:

` https://repo1.maven.org/maven2/edu/berkeley/cs/chisel_2.11/latest.release/chisel_2.11-latest.release.pom  not found:`

I found that https://repo1.maven.org/maven2/edu/berkeley/cs/chisel_2.11/ there is no latest.release. So I just select current latest version 2.2.36, rather than latest.release, as follow:

`libraryDependencies += "edu.berkeley.cs" %% "chisel" % "2.2.36"`

It works.

ucbjrl, 2016-11-17:
If you add the following to your build.sbt:

    resolvers ++= Seq(
      Resolver.sonatypeRepo("snapshots"),
      Resolver.sonatypeRepo("releases")
    )

You should find that `"latest.release"` works as expected.

I've removed the "Getting Started" section of the README. It was out of date and we encourage people that are really getting started to start with [Chisel3](https://github.com/ucb-bar/chisel3).

closed 2016-11-17 by ucbjrl

colin4124, 2016-11-18:
@ucbjrl Thinks very much. I want to use Chisel3 too, but there is still some project such as rocc-template still used Chisel2. I want to use ROCC, do you have some resources about ROCC besides rocc-template?