ucb-bar/chisel2-deprecated #745: documentation link is dead

PeterAaser, 2017-08-21:
Links to documentation repo is dead :(

shunshou, 2017-08-21:
Hmm... looks like you're not the first to think ucb-bar/chisel is still being actively worked on. We probably need to have that in bigger letters somewhere...

The new and improved (and actively in development) version of Chisel sits @ https://github.com/freechipsproject/chisel3

There's quite a bit of documentation @ https://github.com/freechipsproject/chisel3/wiki

(and more to come).

PeterAaser, 2017-08-21:
I'd love to use chisel 3!
Sadly, it's for a course, and none of the teachers have used chisel 3, and we have a lot of quite advanced chisel 2 libraries that we use. While I don't expect any feature requests or issues to be followed up it would be pretty nice to just have the tutorial docs. I did find them in a fork so it's not a pressing issue, but it would be handy to have them here :)

edwardcwang, 2017-08-21:
I'm interested to hear that you're using it for a class. Is the course from the https://en.wikipedia.org/wiki/Norwegian_University_of_Science_and_Technology by any chance? I'd just be curious to see what the class is about./if you have a course website, and how you're using Chisel in it.

PeterAaser, 2017-08-21:
It's for the "computer design project" where students form groups of 7-8 people and basically create a computer from scratch. I attended two years ago, and we created a chip that took in video via HDMI, performed convolution on the videoframes on an FPGA programmed with chisel, and outputted the result on another HDMI port. We even had 8 layer PCBs we designed printed in china, and it also came with an ARM cortex IO processor for flashing and programming the convolution accelerator. The course is worth twice that of a regular course (and quadruple the effort!!)

ucbjrl, 2017-08-21:
@PeterAaser, could you point out the dead links?

PeterAaser, 2017-08-21:
@ucbjrl the one linking to docs:
https://github.com/ucb-bar/chisel-doc

ucbjrl, 2017-08-21:
That's a private repo. Where did you see the link to it?

PeterAaser, 2017-08-21:
@ucbjrl it's in the readme under the Documentation header.

schoeberl, 2017-08-25:
As I was recently teaching a Chisel tutorial and for various reasons I’m still using version 2. Therefore, I’ve “rescued” the Chisel 2 documentation for the tutorial:

https://github.com/schoeberl/chisel2-doc <https://github.com/schoeberl/chisel2-doc>

Cheers,
Martin

abhishek42, 2017-11-16:
The link to chisel 2.2 tutorials at https://chisel.eecs.berkeley.edu/2.2.0/getting-started.html takes to the tutorials written for Chisel3. If i am not wrong the tutorials will differ for both chisel 2.2 and chisel3 , right? 

The directory structure mentioned on the link above for chisel tutorials is different from what is used in chisel3 tutorials. I am some project which was built on chisel2.2 and hence need to learn it. If anyone has a link to chisel2.2 tutorials, it would help out people starting with it.

@schoeberl By any chance, did you rescue chisel2.2 tutorials as well?

chick, 2017-11-16:
I think the problem you are describing is that this [chisel-tutorial](https://github.com/ucb-bar/chisel-tutorial) link takes you to the latest branch which is chisel3.  The chisel2 code is still there under [chisel2 branch](https://github.com/ucb-bar/chisel-tutorial/tree/chisel2).  There probably ought to be a separate release tagged.

abhishek42, 2017-11-17:
ya. Apologies. I found it after posting it here.

closed 2021-07-09 by PeterAaser